JMRI/JMRI #13872: LCC Configure window title bar off screen and which window is on top can be wrong

sometimegithubuser, 2025-02-17:
LCC Configuration windows are often launched with the title bar off the top of the screen. If I maximize them I can get the title bar back and it does not stay. When I select a different config window, others repaint and frequently revert to the title bar being off the top of the screen.

It does not matter if I launch the config windows from Open LCCPro or from the Legacy Configure Nodes windows. 

I've tried editing the user-interface.xml, even deleting it. That fixes a window or two, and then they start reverting to off-screen. 

Additionally, the more windows I have open the more likely they are to get confused about what's on top so minimizing one causes several to minimize or to just hang the whole set of windows so I have to kill -kill the Java process. 

JMRI 5.11.3.Rc13cc39a93b 
Raspberry Pi 5 8G
"Debian GNU/Linux 12 (bookworm)" fully patched
Java openjdk 17.0.14 2025-01-21

sometimegithubuser, 2025-02-17:
[profile.zip](https://github.com/user-attachments/files/18817595/profile.zip)
[session.log](https://github.com/user-attachments/files/18817596/session.log)
[systemInfo.txt](https://github.com/user-attachments/files/18817597/systemInfo.txt)

crosland, 2025-02-17:
I have seen, and reported, similar issues with R-Pi but I don't think there is any easy fix in JMRI.

Do you have the status bar at the top of the screen? I seem to recall this being the root of the issue with the OS not correctly reporting the extents of the visible screen.

I just manually move the window. The new position is then saved in JMRI's settings.

sometimegithubuser, 2025-02-18:
The 1st and 2nd Config windows are fine. After the windows usually have the title bar off the top. I can't move them unless there is a trick using keys and cursor. And they go back to the top even when I do move them by minimizing them, to see the task bar, then maximizing them and then re-sizing them. 

And then issue with what's on top is a problem too. Once there are several windows open then seem to become groups so minimizing one minimizes them all.

crosland, 2025-02-18:
See issue #12692

To move a window, hold down the ALT key and click in the window and drag it.

Pugwash1, 2025-02-18:
Make sure you monitor is pluged in and powered on.
or
Use a hdmi dongle
or
tell the pi theres no screen attached and to pretend its.....

Try and stick with GTK+, its native linux, understands wayland a lot better than other.

Steve G.

sometimegithubuser, 2025-02-19:
Thank you Alt-click window move sequence. I've moved the menu bar from the top to the bottom of the screen. That did not solve the title bar issue at all. The 3rd and 4th Config windows still opened with the title bar off the top of the screen. Using Alt-click, I could bring them down though. So it's better. And the minimization issue is still present. I have five Config windows open and I just clicked the minimize button on one and ALL JMRI windows minimized, including Decoder Pro and Open LCC pro windows.

sometimegithubuser, 2025-02-19:
> Make sure you monitor is pluged in and powered on. or Use a hdmi dongle or tell the pi theres no screen attached and to pretend its.....
> 
> Try and stick with GTK+, its native linux, understands wayland a lot better than other.
> 
> Steve G.

Did the comment get put under the correct issue? Seems to be unrelated.

Pugwash1, 2025-02-19:
Rasp pi now uses wayland by default rather than X, first they used wayfire and then labwc. Wayland itself has rules about overlapping windows, or windows that set their own position.  The pi composters, first wayfire and now labwc don't deal with the problem  well. In addition  the vnc / labwc interface has problems when there is no active hdmi connection and no directions given to tell it how to behave.
If you are using vnc or tigervnc or remains vnc when you connect to a pi running wayland and either of their composters all hell breaks loose. To minimize the effects, it does not  cure them totally, use a real monitor plugged in and turned on, an hdmi dongle, or tell the pi what the video should be thru the use of /boot/firmware/cmdline.txt (this used to be in config.txt before bookworm)
Gtk+ gui uses gtk3 which is aware of some of the issues.

sometimegithubuser, 2025-02-19:
I am sitting front of the Pi using an HDMI connected monitor, keyboard and mouse, not remote access.  

The overlapping windows, which one is on top, allowing me to bring the window of my choice to the front, and minimizing only one windows are all affected by the bug. Makes it very hard to use. 

Does Windows have any of these issues or are they specific to the Pi or, more generally, Linux? Is there hope of a fix for the Pi or should I give in and buy a Windows machine to use instead?

Pugwash1, 2025-02-19:
Turn off wayland.

sudo raspi-config
Then Advanced  -> Wayland -> X11 and follow the prompts.

If you do need to remote in vi vnc you need to configure that again before rebooting.

sometimegithubuser, 2025-02-19:
Thank for the suggestion to turn off Wayland and switch to X11. That made a huge difference. I still need to do some testing, but it's clearly a lot better!

sometimegithubuser, 2025-02-22:
So far everything is working, even with many windows open. Too bad that Wayland does not work as it's the default.

closed 2025-02-22 by sometimegithubuser