JMRI/JMRI #13892: Dark Mode

rickbeaber, 2025-02-26:
Has there been any discussion about adding a dark/high contrast mode to JMRI, especially in the operations section.  It would be very helpful for those of us whose vision is not as good as it used to be.  Many other pieces of software are adding this.

Thank you,

-Rick Beaber

bobjacobsen, 2025-02-26:
Interesting idea!

In another project, we used the [Darklaf look and feel](https://github.com/weisJ/darklaf) to do this.  It's not as polished as the vendor-provided look and feels, so there are some tradeoffs in using it. It might be worth investigating that approach.

bobjacobsen, 2025-02-26:
I'll see if Darklaf will work with JMRI as a test.

bobjacobsen, 2025-02-27:
@rickbeaber There are installers with DarkLAF High Contrast (a dark "Look & Feel") enabled by default at:

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/209/

Note that these installers are _current_ JMRI code (e.g. 5.11.3) and _only_ do dark mode. 

Could you please install one of these and try it out?  I'm curious how well this works for your panels, etc.

If you don't like it, you can reinstall JMRI 5.10 or later right on top of this to get back to the normal look and feel.

Thanks in advance!

KenC57, 2025-02-27:
I'll say it makes the normal screens rather dark but high contrast.

 

But on one panel I tried, almost all of the text boxes showed as black
with black text it seems. The background of that layout was showing
white or the normal gray. 

 

Another layout the panel came up all black background and could see
signals and sensors but the track and most normal text was black on
black. Only some gray text sort of showed up.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org <http://www.jmri.org/> 

www.fingerlakeslivesteamers.org
<http://www.fingerlakeslivesteamers.org/> 

www.cnymod.com <http://www.cnymod.com/>

bobjacobsen, 2025-02-27:
Unfortunately, it would be a _lot_ of work to just have a dark background apply to just certain windows.  E.g. dark tables, but light panels.  To use this dark mode, I think people would have to build panels that are suited to it.  For Layout Editor,  I think you can set the color of the background directly.  For Panel Editor and Control Panel Editor, I think you'd need to create a suitable background image.

rickbeaber, 2025-02-27:
I'm getting this error when I click on any menu item in the new build.
Unrecoverable error encountered.
This application will quit.

class com.github.weisj.darklaf.compatibility.SwingUtil (in unnamed module
@0x7fc2413d) cannot access class sun.awt.SunToolkit (in module
java.desktop) because module java.desktop does not export sun.awt to
unnamed module @0x7fc2413d
java.lang.IllegalAccessError
java.lang.IllegalAccessError: class
com.github.weisj.darklaf.compatibility.SwingUtil (in unnamed module
@0x7fc2413d) cannot access class sun.awt.SunToolkit (in module
java.desktop) because module java.desktop does not export sun.awt to
unnamed module @0x7fc2413d
 at
com.github.weisj.darklaf.compatibility.SwingUtil.grab(SwingUtil.java:109)
 at
com.github.weisj.darklaf.ui.popupmenu.MouseGrabber.grabWindow(MouseGrabber.java:74)
 at
com.github.weisj.darklaf.ui.popupmenu.MouseGrabber.stateChanged(MouseGrabber.java:112)
 at java.desktop/javax.swing.MenuSelectionManager.fireStateChanged(Unknown
Source)
 at java.desktop/javax.swing.MenuSelectionManager.setSelectedPath(Unknown
Source)
 at java.desktop/javax.swing.plaf.basic.BasicMenuUI.appendPath(Unknown
Source)
 at
java.desktop/javax.swing.plaf.basic.BasicMenuUI$Handler.menuDragMouseDragged(Unknown
Source)
 at java.desktop/javax.swing.JMenuItem.fireMenuDragMouseDragged(Unknown
Source)
 at java.desktop/javax.swing.JMenuItem.processMenuDragMouseEvent(Unknown
Source)
 at java.desktop/javax.swing.JMenuItem.processMouseEvent(Unknown Source)
 at java.desktop/javax.swing.MenuSelectionManager.processMouseEvent(Unknown
Source)
 at
com.github.weisj.darklaf.ui.popupmenu.MouseGrabber.eventDispatched(MouseGrabber.java:211)
 at
java.desktop/java.awt.Toolkit$SelectiveAWTEventListener.eventDispatched(Unknown
Source)
 at
java.desktop/java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Unknown
Source)
 at
java.desktop/java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Unknown
Source)
 at java.desktop/java.awt.Toolkit.notifyAWTEventListeners(Unknown Source)
 at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
 at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
 at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
 at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
 at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
 at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
 at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
 at java.base/java.security.AccessController.doPrivileged(Unknown Source)
 at
java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown
Source)
 at
java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown
Source)
 at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
 at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
 at java.base/java.security.AccessController.doPrivileged(Unknown Source)
 at
java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown
Source)
 at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
 at
java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown
Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown
Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown
Source)
 at java.desktop/java.awt.WaitDispatchSupport$2.run(Unknown Source)
 at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
 at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
 at java.base/java.security.AccessController.doPrivileged(Unknown Source)
 at java.desktop/java.awt.WaitDispatchSupport.enter(Unknown Source)
 at java.desktop/java.awt.Dialog.show(Unknown Source)
 at java.desktop/java.awt.Component.show(Unknown Source)
 at java.desktop/java.awt.Component.setVisible(Unknown Source)
 at java.desktop/java.awt.Window.setVisible(Unknown Source)
 at java.desktop/java.awt.Dialog.setVisible(Unknown Source)
 at jmri.util.swing.JmriJOptionPane.displayDialog(JmriJOptionPane.java:284)
 at
jmri.util.swing.JmriJOptionPane.showOptionDialog(JmriJOptionPane.java:168)
 at
jmri.util.swing.JmriJOptionPane.showMessageDialog(JmriJOptionPane.java:75)
 at
jmri.util.swing.ExceptionDisplayFrame.displayExceptionDisplayFrame(ExceptionDisplayFrame.java:41)
 at
jmri.util.exceptionhandler.UncaughtExceptionHandler.uncaughtException(UncaughtExceptionHandler.java:40)
 at java.base/java.lang.ThreadGroup.uncaughtException(Unknown Source)
 at java.base/java.lang.ThreadGroup.uncaughtException(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.processException(Unknown
Source)
 at
java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown
Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown
Source)
 at
java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown
Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

I'm running the Eclipse Java.  Do I need to be running Oracle's Java?

Thank you,

-Rick

bobjacobsen, 2025-02-27:
Which Java version number are you using?  I've only seen errors like that with greater than Java 18

rickbeaber, 2025-02-28:
Eclipse Temurin 17.0.14+7

Thank you,

-Rick

bobjacobsen, 2025-02-28:
That's interesting!  There's an [issue filed against DarkLAF](https://github.com/weisJ/darklaf/issues/343) from November 2023 that describes this.  Apparently they're "going to fix it in the next version", which hasn't come out yet.  Unfortunately, they way they're planning to fix this is to _require_ Java 17 to use it.  JMRI is still only requiring Java 11, recommending 17.  It's likely to be a while before (a) they fix this and (b) JMRI moves forward to require Java 17.

You could try a Java 11 JRE if you want to see how this looks.  I'm not sure that's worth the effort, because I don't think we can use this right now.

Sorry this approach is not working out. I'll look around to see what else I can find.

rickbeaber, 2025-02-28:
> There's a PR in another project
> <https://github.com/openrocket/openrocket/pull/2261> that says they've
> developed a workaround.

Bob:
     Tried the workaround, but maybe I'm not doing it correctly.  It just
returns the build number, but doesn't run JMRI.

I ran at a command prompt:
java -jar --add-exports java.desktop/sun.awt=ALL-UNNAMED jmri.jar

Thank you,
-Rick

bobjacobsen, 2025-02-28:
Which type of computer do you have:  Windows, Linux, Mac?

rickbeaber, 2025-03-01:
Windows 11.

Thank you,

-Rick

bobjacobsen, 2025-03-02:
Sorry for the delay.  It took a little bit of time to figure out you to get that option into the normal JMRI startup.

Could you please:
 - Find a "jmri.conf" file in your JMRI program directory (where the program is installed)
 - Make a copy of that as a back up and put it somewhere else on the computer
 - Edit that file and add the following line at the bottom. Note that the quote marks should be included; that's exactly the line to add.

```
default_options="-J--add-exports=java.desktop/sun.awt=ALL-UNNAMED"
```

 - Start JMRI and see if that bypasses the error.

Thanks, and sorry for the trouble!

klk32003, 2025-03-02:
Don't forget to copy the "jmri.conf" file into your "settings location" (C:\Users\your_user_name\JMRI\) before you edit the line at the bottom.

rickbeaber, 2025-03-03:
klk32003, copied the file to the prefs folder like you said.

Bob,
Unfortunately, I now get another error:
Unrecoverable error encountered.
This application will quit.

class com.github.weisj.darklaf.compatibility.SwingUtil (in unnamed module
@0x7fc2413d) cannot access class sun.awt.SunToolkit (in module
java.desktop) because module java.desktop does not export sun.awt to
unnamed module @0x7fc2413d
java.lang.IllegalAccessError
java.lang.IllegalAccessError: class
com.github.weisj.darklaf.compatibility.SwingUtil (in unnamed module
@0x7fc2413d) cannot access class sun.awt.SunToolkit (in module
java.desktop) because module java.desktop does not export sun.awt to
unnamed module @0x7fc241

Thank you,

-Rick

bobjacobsen, 2025-03-03:
Sorry this is so hard to check out.  That's the same underlying error,  which indicates that the change to jmri.conf did not do what it's supposed to.  Did you add that line to the file after you copied it?

rickbeaber, 2025-03-04:
> Bob,
>      No need to apologize.  I've spent most of my life in IT, remote
> troubleshooting is hard.
>
> I gave up on the machine I kept getting the error on and tried another
> machine.  It works great.  Much easier on the eyes.  I didn't see any real
> issues in the operations portion.  Hopefully, there's a way that it can at
> least be an option going forward.  Maybe with a warning that panels may
> have to be edited when using this.
>
> Thank you very much for doing this.  It is much appreciated.
>
> -Rick

Sorry, forgot to add it works fine with the workaround.  Still got the
error on the new machine without the default options change.

-Rick

bobjacobsen, 2025-03-04:
Thanks!  I'll make it an option pretty soon, though maybe not in time for Saturday's test release.

bobjacobsen, 2025-03-08:
@rickbeaber 

I've got a build with DarkLAF properly integrated now.   If you have a chance, could you remove that work-around and then try an installer from

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/

Thanks!

rickbeaber, 2025-03-09:
Bob:
     Unfortunately, it will not open.  Installed over the previous build
you sent me the link to.  It does nothing when clicking on either icon.
Running installtest passes with no errors.  Anyplace else I can look to
help you troubleshoot it?

Thank you,

-Rick.

bobjacobsen, 2025-03-11:
@rickbeaber Could you try the installer at 

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/215/

when you have a chance?  @klk32003 has been helping with a new launcher and we think this build might do it.

Thank you!