JMRI/JMRI #13896: Running JMRI on Windows 11

ErniePuddick, 2025-03-01:
I'm new to this forum so if this is in the wrong place just tell me.

I've had a Sprog for a long time.  At first it worked fine the required downloads worked fine.  Other things meant I did not get use the Sprog and my Laptop changed.  With Windows11 I've struggled to get Java loaded and likewise JMRI..
My Sprog has been upgraded, which has spurred me on to get it running.
I've managed to get Java installed but failed with JMRI.  It didn't want to download the file, but that was solved by using tips on the JMRI website.  When I try and open it I get "This App can't run on your PC"...

Any suggestions are appreciated.

Thanks

Ernie Puddick

[image]

PUlvestad, 2025-03-03:
Your file shows 0 KB, meaning there isn't anything there. The downloaded file should be about 182 MB.
I recommend you join the JMRI users group for help. https://groups.io/g/jmriusers

ErniePuddick, 2025-03-03:
Good morning,I’m still waiting for my subscription to be approved.ThanksErnie PuddickSent from my iPad

closed 2025-05-26 by github-actions[bot]