JMRI/JMRI #13901: Update Pi4J from v1.2 to v2.7 or later

mstevetodd, 2025-03-06:
The RaspberryPi GPIO connection no longer works with recent RPi OS and hardware changes. The Pi4J v1.2 library used by the connection is no longer supported or working. The current version 2.7+ is a complete rewrite and redesign. So the change to the JMRI RPi connection is much more than just replacing the .jar files. It will need to be redesigned to work with the new version.

Note: I took a look at the changes, but I'm not familiar enough with the existing code to understand what needs to be replace with what. It is my understanding that @pabender did the original work. It would be great if he was available to make this change, or at least provide expertise on what needs to be replaced with what.

pabender, 2025-03-07:
I’ll take a look at it, but it may be a few weeks before I can get to it.

Paul

danielb987, 2026-04-08:
See this comment: https://github.com/Pi4J/pi4j/issues/214#issuecomment-1114879271

It seems that sudo is not needed:
https://github.com/Pi4J/pi4j/issues/212#issuecomment-2354789391
https://www.pi4j.com/documentation/providers/
https://www.pi4j.com/documentation/providers/ffm/

pabender, 2026-04-14:
@mstevetodd,

I figured out what the missing library is… and it’s a problem with how you and I both updated the software…

My process was this…

1. Download the latest test release to the Pi
2. compile on my laptop the jar from my branch
3. push the jar into the proper location.

The missing step there is that this is a new version of the library, so we also needed to push the pi4j jars up from the lib directory of the repo.  

After that doing that, it was discovered that there was a missing runtime dependency for one of the plugin jars… and a missing native library that needed to be installed.

I’m still working out one kink that prevents initialization of an IO pin, but I do now have it loading…

I’m going to work through that before I push another update to my branch.