JMRI/JMRI #13903: DCC++ Traffic Monitor window checkboxes disappear

jdenm8, 2025-03-07:
Hi, 
When the DCC++ Traffic Monitor window is narrowed, the checkboxes along the bottom disappear. Re-widening the window does make the checkboxes re-appear. 

However, if the window is closed while it's narrow, and then re-opened, the checkboxes won't re-appear again until the window is widened, closed, then reopened again.

Window at minimum width before checkboxes disappear
[image]

Window at maximum width before checkboxes reappear
[image]

Re-opened and expanded window after closing the window with the checkboxes gone
[image]

Personally tested on:
JMRI 5.6 on Java 11.0.21 running on Windows 10 22H2
JMRI 5.8 on Java 17.0.13 running on SteamOS 3.6.22 (Linux Kernel 6.5.0)

Observed on:
JMRI 5.10 on Java 17.0.13 running on Windows 11

mstevetodd, 2025-03-07:
Note that this issue can be recreated using the DCC++ Simulator connection, so no specific hardware needed.

mstevetodd, 2025-03-07:
The Loconet Monitor handles the resize better by rearranging the checkboxes as the window shrinks. Perhaps this code could be copied over to the DCC++ Monitor.

[image]

Ash-4, 2025-03-07:
Please change the default, so that the 'Show raw data' checkbox is checked and the 'Show translation' is unchecked.

danielb987, 2025-03-07:
The solution is probably to use jmri.util.swing.WrapLayout for the JPanel that holds the checkboxes.

closed 2025-06-02 by github-actions[bot]