JMRI/JMRI #13904: Soundtraxx Tsunami 2

brownchristopher1969, 2025-03-08:
On a eight function Soundtraxx Tsunami 2 decoder, Decoder Pro does not have CV address 1.263 & 1.264.  I notice this because I have several engines configured the same way and I just duplicate an ID, give it a new number and Change the "Active Address" and then let it write all sheets.  CV1.263 & 1.264 control functions 7 & 8.  Will those two CV's be added soon to a new revision?  I am mapping function 8 to 0.  On the Rapido GP38, this allows the guage lights on the stand to come on when the function 0 is on.
 
Thank you very much for all your hard work in creating this software.
Best regards,
Christopher Brown

PUlvestad, 2025-03-09:
Those CV's are in the main definition. What version of JMRI are you running?

What decoder does it point to when you click on New Locomotive and let JMRI try to find the appropriate definition?

Peter

PUlvestad, 2025-03-09:
@mmosher5501

mmosher5501, 2025-03-09:
What decoder did you pick?  For a Rapido GP38, the best choice would be TSU-21PNEM8, which in JMRI 5.11.2 does have CV's 1.263 & 1.264 defined, 8 function support was added 2020-07-27, so I think all 5.xx.x should have it.

brownchristopher1969, 2025-03-10:
I have version 5.10.  Isn't that the latest?  I am not a fan of Loksound.  I strip it all out and install Soundtraxx PNP8.

mmosher5501, 2025-03-13:
For me in 5.10
All 5 of:
Tsunami2 Diesel
     TSU-PNP8 EMD Diesels
     TSU-PNP8 EMD-2 Diesels
     TSU-PNP8 GE Diesels
     TSU-PNP8 Alco Diesels
     TSU-PNP8 Baldwin Diesels
Have the var/CV's for FX7 Effect (CV 16.1.263 & CV 16.1.391) and FX8 Effect CV 16.1.264 & CV 16.1.392) on the Function Map pane.
Perhaps you have old versions of the TSU2 files in your 
<profile folder>\decoders
and/or
<profile folder>\xml\decoders\soundtraxx
folders

brownchristopher1969, 2025-03-13:
Under "Function Map" it only shows- 
FX Headlight, Backup light
FX3
FX4
FX5
FX6
Under "CV's", it has up to CV 1.262 (mapping for function FX6 effect) then jumps straight to CV 1.273.  I need CV's 1.263 & 1.264 to map functions 27 & 28.
Under "About Decoder" I see version 5.10
Can you attach a photo/screen shot of your CV's page?  I'd like to see CV 1.263 & 1.264 listed
Thank you

PUlvestad, 2025-03-13:
[image]

PUlvestad, 2025-03-13:
As Micheal asked, do you have an old definition in your users folder? 
Are you using the TSU-PNP8 definition or the TSU-PNP definition?

[image]

brownchristopher1969, 2025-03-13:
PNP8.  What is an "old definition"?  Sorry not familar with that.  Your screen shots are what I want mine to be like.
Thank you very much!

mmosher5501, 2025-03-14:
If you CV tab lists the CVs as 1.257 instead of 16.1.257 then you are using an old definition.
In DP, pick help File Locations

[image]

Then pick Open Profile Location, then in your computer's file explorer look for a sub folder called decoders, if any files named Soundtraxx..., delete them or move them to some temp folder.

Also check the xml\decoders\soundtraxx folder and delete or move any files there.

Any files in those locations will be used in place of the same name files that get installed in the JMRI program locations.

mmosher5501, 2025-03-14:
> Mine does show the format 16.1.257. I cut & pasted all soundtraxx decoder files to a temp folder. I checked in xml and there is no soundtraxx files there. Now when I try to double click on row nothing happens. I can't get in to see or set CV's. Are you using 5.10 or 5.10.2?

You’ll need to pick actions - rebuild decoder index after moving the files out of that folder

Michael Mosher 
Member SFRH&MS www.atsfrr.net/

brownchristopher1969, 2025-03-14:
Ok, did that.  Now I can open a row but it's the same thing, goes from 16.1.262 straight to 16.1.273 as before.

brownchristopher1969, 2025-03-14:
They are all under TSU-2200 is why they don't have anything pertaining to functions 27 & 28.  Thank you all very much for your time and suggestions.  It is very much appreciated.

brownchristopher1969, 2025-03-14:
I know now.  I got so use to duplicating the previous entries I never stopped to think until I started using the PNP8's.  
Thank you all once again for taking  the time to help me.

closed 2025-03-27 by brownchristopher1969