JMRI/JMRI #13910: LogixNG Show Dialog Action

dsand47, 2025-03-11:
When the **Multiline** option is used, the buttons can be outside of the displayed window.

[image]

The bottom of the window needs to dragged down to reveal the buttons.

danielb987, 2025-03-13:
@dsand47 
Could you try adding `_dialog.revalidate()` before or after line 294 in `jmri.jmrit.logixng.actions.ShowDialog` to see if that helps?

https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/logixng/actions/ShowDialog.java#L294

The difference between singe line and multi line is that the latter uses HTML to show multi lines in a JLabel. And maybe that doesn't work so well. It might be dependent of which L&F you use, but I'm not sure. But my idea is that maybe the dialog must first be made visible and then revalidated for the dialog to know the size of the HTML. But it's a guess.

dsand47, 2025-03-13:
That did not fix the problem.  The actual problem appears to be that the window is narrow and lines wrap.  It is also strange that the content is inset almost half of the space.

Changing the main panel from BoxLayout to BorderLayout resolves the line wrapping and the inset issues.

[image]

danielb987, 2025-03-13:
Thank you.
Can you create a PR for the change now when you have the solution for the problem?

closed 2025-03-14 by danielb987