JMRI/JMRI #13916: WARN messaged for Sensor userName when EngineDriver opens Control Panel with Indicator icons

Cliff-Anderson, 2025-03-12:
This is only a nuisance, but the cumulative impact of each Indicator icon can really clutter up the Console Window and the log messages for complicated Control Panels that are opened from EngineDriver. 

The specific warning message for a single Indicator Track icon looks like:
"2025-03-12T15:27:57,097 json.JsonNamedBeanSocketService       WARN  - get request for sensor made with user name "Artificial Sensor"; should use system name [WebServer-86]"

Each Indicator Track icon and each Indicator Turnout icon contributes it's own WARN message each time the panel is linked.  

In a club environment with multiple panels and multiple smartphone connections any unnecessary WARN message is a potential distraction from real problems.

The config profile folder with an extremely simplified example that generated the above message is attached.  

LocoNet and LCC Sensors can cause this action.  Maybe other connection types too?

The warning only appears when the EngineDriver (or perhaps WiThrottle?) option for "Web" selects the panel.

The complete session.log file is also attached.

[Sensor_Name_Question.jmri.zip](https://github.com/user-attachments/files/19218882/Sensor_Name_Question.jmri.zip)

[session.log](https://github.com/user-attachments/files/19218884/session.log)

mstevetodd, 2025-03-13:
EngineDriver is simply providing a wrapper for a web browser, so the warning is likely coming from panel.js (the client-side web panel). EngineDriver has no visibility or control of the web content.

panel.js simply requests and processes the xml definition for the panel, which contains userNames in several places. Nothing the web page can do except request json updates of what it is given. However, panel.js does "upgrade" each widget to use the systemName as each is (first) returned by the json server. So, you should only see the WARN once for each widget, as the page loads, not every time the widget changes.
Someone could change the xml storage code in each element type that has the WARN, or (perhaps) override each instance on-the-fly when sent via the web server.

For the end user, I recommend simply ignoring them, or changing the default_lcf.xml to suppress the WARN for JsonNamedBeanSocketService.

--SteveT

Cliff-Anderson, 2025-03-13:
Thanks @icklesteve for the insiders information.  Most JMRI users would not know or even care what the Json programmers rules are.  

The source code you referenced, at line 57 appears to know what it is supposed to do with userName identifiers.  Or as a long time ago retired programmer with multiple forgotten programming languages who has not learned either Java or Json, that is what that line of code looks like to me.  

It is not my place to ask, "Why," but it is unlikely that the explanation would be understandable to me.

Apparently our Control Panels should continue to avoid the use of Indicator icons, as we have been doing for many years.  The Control Panels are designed to incorporate OBlocks and our layout expansion is in the process of implementing the track plan developed on AnyRail and exported to a Layout Editor with identified Blocks, not OBlocks.  

The Item Palette dialog for Indicator Track and for Indicator Turnout expect a field named Control Circuit to be typed or else cut-and-paste to fill in the blank.  The choice of a Sensor seems to be acceptable, but an OBlock seemes to be preferred.  

Cut-and-paste is not available for systemName identifiers from within an active layout.  

All of the other dialog tabs provide pull-down lists and accept selecting a userName.  Since we are using LCC for the expansion, it would be necessary to use a text editor to obtain entries like "MS02.01.57.40.00.37.00.7E;02.01.57.40.00.37.00.7f" instead of "Keddie: West Ladder" to fill in the blank.  It seems unlikely that some future member would know how to decipher the hundreds of such entries when maintaining this layout ten years from now.  

At my age, it is unlikely that I will see this implementation completed.  

There was __NO__ intent to cast any blame on EngineDriver or at @mstevetodd, but only to provide a simple way for some developer to reproduce the WARN message.

dsand47, 2025-03-13:
Cliff,

As of JMRI 5.11.4, the standard JMRI tables now support "Copy System Name" in the right click menu.  Note: OBlocks is not a standard table.