JMRI/JMRI #13955: RailCom feedback from EcoSDetector

maciekish, 2025-03-21:
Hi, i’m unable to find a straight answer to these anywhere and not ready to drop $650 on a command station without knowing. Could someone help me out please?

1) Does JMRI support reporting via ESU ECoS & **ECoSDetector**s? Actual train ID’s, not only plain block detection. 
2) Does JMRI support signal head control via **ECoSDetector Extension**? (32 outputs). 

Thank you

bobjacobsen, 2025-03-22:
You will probably get more responses to this on the JMRI users group mailing list:

https://groups.io/g/jmriusers

kjlisby, 2025-04-04:
I own an ECoS. And I am using it with JMRI. But I don't own any ECoSDetector. I use S88. But that does not prohibit me from tracking my trains. I am running my trains via Signal Controlled Warrants. And that makes JMRI know which train that is where on my layout and to show that information in a panel: 

[image]

However, looking in JMRI\java\src\jmri\jmrix\ecos\EcosSensorManager.java, it seems that ECoSDetector and RailCom is supported by JMRI.

I am not sure what your question number 2 means. You can contact me directly, if you are interested. You can find my e-mail address here: https://lisby.dk/wordpress/  And you can read about how I use ECoS, JMRI etc. to control my trains.