JMRI/JMRI #13959: Signal Mast "Edit" function causes "ERROR - About to set mast to null, which shouldn't happen."

Cliff-Anderson, 2025-03-23:
The attached files show the problem reproduced for a Basic Enhanced type Signal Mast using LocoNet Turnout addresses to control the aspects.

This is the simplest example that I am able to construct.  Even though version 5.11.4 was used, the ERROR messages can be reproduced with versions 5.6, 5.8, and 5.10.  

The discovery was made on a complicated layout, while attempting to edit a WPRR-1971 type Signal Mast using LCC EventID numbers to control the aspects.  Numerous other Signal Mast types with LCC EventID choices caused the same results.  I am not familiar enough with Signal Mast options to give much more help.  I am unable to create Signal Head objects with LCC to compare results.

It is not clear that the reported ERROR is actually causing a real problem to our layout development,  but there is a very heavy investment in time and money and members' commitments involved.  

This provides a risk that I am unwilling to impose on the club, just find a "work-around." 

In the attached log, the first error message occurs at 2025-03-23T12:51:44,349 when the "Add" button was clicked, but the reason given at that stage is not an actual error.

Several minutes later, after the only Signal Mast has been created, and the "Edit" button was clicked, the lengthy action with traceback info was logged.

Nothing was actually changed in the edit mode in the provided example, the "Cancel" button was clicked.

Thank you

[session.log](https://github.com/user-attachments/files/19411149/session.log)

[1_signal_mast.jmri.zip](https://github.com/user-attachments/files/19411151/1_signal_mast.jmri.zip)

dsand47, 2025-03-23:
Cliff,

The signal mast in the xml file refers to signal system "basic-enhanced-A".  The correct name is "basic-enhanced".  If I remove the -A in the xml file, it loads without errors.

Perhaps you have a custom version in the user files location.

The JMRI install at C:\Program Files (x86)\JMRI\xml\signals should have a "basic-enhanced" directory.

These errors are most likely different than the LCC errors.

Dave Sand

Cliff-Anderson, 2025-03-23:
Thank you Dave,

It took some time to find a subfolder named "xml.signals" in my users folder, but removing it and then editing the PanelPro xml file to delete the "-A" removed the problem of strange ERROR messages.  

That subfolder was dated more than five years ago and the reason for it's existence is long forgotten.  Also the text "Signal System Not found for userName null" led me to hunt for missing userName occurrences but not for erroneous systemName sources.  

My apologies to the developers.

closed 2025-03-23 by Cliff-Anderson