JMRI/JMRI #13975: Add CV's

brownchristopher1969, 2025-03-27:
I am really impressed with Decoder Pro and it's approach to make settings easier.  I currently use it to back-up all my loco's.  Here the problem I see.  People get so reliant on Decoder Pro, when many people are running at a layout and a CV needs to be changed, they feel the need to "run" to Decoder Pro to make an adjustment when they could make the adjustment on a program track or program right on the track in "op" programming.  The largest disconnect I see is the majority of Hobbists running loco's equipped with DCC have no clue what CV's do what.  I personally only use Decoder Pro's "CV" tab.  And I love how hovering over a CV or under any other tab gives a desciption as to what that setting does, awesome! 
I humbly request one change that could help the growing gap in the DCC community.
Either add the CV address next to each setting or add the CV (perhaps in brackets) at the end of the hover over, pop up description.  In doing this, it does not change the software in a way that effects the Decoder Pro user BUT it does give the users the opportunity to note the CV in question so the next time they want to make a change at the layout, one can simply refer to a note they wrote or "learned".  I strongly believe becoming comfortable setting CV's outside Decoder Pro is an important part of the hobby.  From my experience of learning through the manuals, I can say I feel much more comfortable running at layouts and being able to tweak settings on the fly without having to rely on Decoder Pro 100%.  I believe what I am suggesting is a positive opportunity for the end user to grow in knowledge without forcing a change if one does not care to learn CV's.  Do I have them all memorized, nope, but I always have a manual with me for myself and others.  Setting CV's at a layout should not be a scary thing but unfortunatly I find a lot of people running are actually afraid to set CV's in fear they will mess something up.  Suppose they do, so what.  When they access their Decoder Pro file for that loco they can re-write it and hopefully learn for the experience.  Does this make sense?  I'm curious what users think.  It doesn't affect me either way but I see great poetential for those wanting to feel more confident in understanding CV's.
Thank you so very much for your time (in reading this and creating the software!)

Christopher Brown

PUlvestad, 2025-03-27:
>>The largest disconnect I see is the majority of Hobbyists running loco's equipped with DCC have no clue what CV's do what.

That's the point of DecoderPro, so people don't need to know what CV does what.

Peter

brownchristopher1969, 2025-03-27:
> > > The largest disconnect I see is the majority of Hobbyists running loco's equipped with DCC have no clue what CV's do what.
> 
> That's the point of DecoderPro, so people don't need to know what CV does what.
> 
> Peter

Is there any harm in having it is the way it is plus showing what CV your setting?

bobjacobsen, 2025-03-27:
Have you tried Settings -> Roster -> Programmer -> Show CV Numbers in tooltips?

brownchristopher1969, 2025-03-27:
> Have you tried Settings -> Roster -> Programmer -> Show CV Numbers in tooltips?

Under my settings, I don't see "Roster"

[image]

bobjacobsen, 2025-03-27:
I think maybe it's called "Preferences" on Windows.  Look under the File menu.

https://www.jmri.org/help/en/package/apps/TabbedPreferences.shtml

brownchristopher1969, 2025-03-27:
> I think maybe it's called "Preferences" on Windows. Look under the File menu.
> 
> https://www.jmri.org/help/en/package/apps/TabbedPreferences.shtml

It is there and it does work!  Thank you!

closed 2025-03-27 by brownchristopher1969