JMRI/JMRI #13985: New Dark mode is showing in 5.9.7 release notes

PUlvestad, 2025-03-28:
The new Dark mode is showing in 5.9.7 release notes when it just got added 2 weeks ago.
https://www.jmri.org/releasenotes/jmri5.9.7.shtml
@bobjacobsen 

I looked at code for that page and don't see it.

Peter

bobjacobsen, 2025-03-28:
Fixed.

bobjacobsen, 2025-03-28:
While developing a release, we include the list of changes from a file in the JMRI/JMRI repository.  We then replace that with the contents as part of the release build. I erroneously left that include in the 5.9.7 release note, so it was showing the most recent changes too.

closed 2025-03-28 by bobjacobsen

PUlvestad, 2025-03-28:
I saw line 308, but it looked to be commented out. Isn't that what the  does?

PUlvestad, 2025-03-28:
I figured it out. That line is hidden but the "include" still runs.

Thanks Bob