JMRI/JMRI #13991: Very slow behavior and many 306 errors with DecoderPro connected to Intellibox 2Neo

Petit-Barde, 2025-03-30:
Hello,

I installed the DecoderPro version 5.10 and connected it to an Intellibox 2Neo via USB.
The 2Neo is similar to Intellibox II but with mainly WLAN and Integrated LocoNet-over-TCP server for all control systems with LocoNet-over-TCP. For example JMRI and Rocrail.
In Edit / Preferences ... / Connections : 
 -  System manufacturer : Uhlenbrock
 -  System connection : Intellibox II
 - Settings : 
     - serial port : COM1
     - Command station type : Intellibox II
     - Connection prefix : L
     - Connection name : Loconet
     - Baud rate : 115,200 bps
     - Serial connection uses : hardware flow control (recommended)
     - Interrogate Sensors/Turnouts On Start : Yes
     - Expanded Protocol (XP Slots) : No
     - Packetizer type : Normal (recommended)
     - Transponding present : No
     - Turnout command handling : Normal
     - Output interval (ms) : 250
DecoderPro sees the Intellibox 2Neo (the layout power state changes from Unknown to On).
When pushing the layout power button the first time, the change on the Intellibox 2Neo appears immediately, but when pushing the button again, a delay of some seconds appears before the Intellibox reacts and this delay increases each time I push the Layout power button.
I have to wait several seconds before pushing again the Layout Power button so that the Intellibox reacts again immediately.
A similar delay problem appears while trying to program a decoder (ESU Locksound V5 micro DCC) : 
   - the Intellibox reacts with more and more delay so that the 306 error comes very often and the CV reading is wrong (CV values appears in the wrong slots, other are not read, etc.). This behaviour appears for example when trying to read the CVs data in the Basic tab of the Program dialog, but also in the CVs tab where most of the read CVs are wrong and require several readings to be read correctly.
   - The intellibox continues to read and send the requested data (which is not actually filed in the DecoderPro dialog) even many seconds after DecoderPro shows it gathered all the data (by showing OK on the bottom of the Basic tab for example).

When I swap from DecoderPro to RocRail with exactly the same layout and PC and connection configuration the Intellibox reacts always immediatly and I have no problem to configure the same decoder for example by reading at once the 28 CV67 - CV94 values of the loco speed profile (no delays, no errors).

Best regards.

bobjacobsen, 2025-03-30:
I think we need some more information on what's happening.

1) Could you please open the LocoNet Monitor (from the LocoNet menu) and check what's happening during these delays?  Is there a lot of traffic while the delay is happening, or is the LocoNet idle then?

2) Open "Monitor LocoNet Stats" from the LocoNet menu, click Update.  What does the window now say?  This should tell us if LocoNet errors are being seen.

3) Finally, open the System Console from the Help menu and see if there are any errors or warnings reported.  If there are, copy and paste them into here.

Thank you!  And sorry for the trouble.

Petit-Barde, 2025-03-30:
Hello BobJacobsen,
I will try to answer to each point : 
1) Loconet Monitor : 
  - the delay appears while no loconet message is observed in the Loconet Monitor.
  - the Intellibox reacts as soon as the message appears in the Loconet Monitor.
  - after a while, when the On/off button is pressed on the DecoderPro, the message appears and the Intellibox reacts immediately.
  - If I press again the button immediately, nothing happens (no loconet message in the monitor, no Intellibox reaction) until almost 9s seconds. After that, the message appears and the Intellibox react both at the same time.

2) Monitor loconet stats : 
  - nothing appears on the loconet stats while playing with the Layout Power button.
  - same thing while trying to read the Basic tab CV data, many 306 errors, most wrong data  in each field, but no monitor stats errors reported after updating the loconet stats dialog.
  - the loco in on the progamming layout, CV are read with Direct byte option, when I request to read the full CVs of the Basic tab, the Intellibox put the loco in program mode (all lights and sounds off).

3) System console : there are 2 warings so I copy and paste the complete trace in the console : 

22:27:24,788 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
22:27:24,812 apps.util.Log4JUtil                   INFO  - This log is stored in file: C:\Users\Marc\JMRI\log\session.log [main]
22:27:24,813 apps.util.Log4JUtil                   INFO  - This log is appended to file: C:\Users\Marc\JMRI\log\messages.log [main]
22:27:24,835 apps.AppsBase                         INFO  - DecoderPro version 5.10+Rca461bd266 starts under Java 17.0.14 on Windows 10 amd64 v10.0 at Sun Mar 30 22:27:24 CEST 2025 [main]
22:27:24,999 apps.gui3.Apps3                       INFO  - Starting with profile Mon_Réseau_JMRI.3e1c065a [main]
22:27:25,298 jmri.util.node.NodeIdentity           INFO  - Using d1481c4e-a1a5-4c77-a97c-6868d2e2b8eb as the JMRI storage identity for profile id 3e1c065a [AWT-EventQueue-0]
22:27:25,378 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]
22:27:25,448 .loconet.uhlenbrock.UhlenbrockAdapter INFO  - Connecting Uhlenbrock via COM1 CP2102 USB to UART Bridge Controller [main]
22:27:25,448 .loconet.locobuffer.LocoBufferAdapter INFO  - LocoNet: Port Silicon Labs CP210x USB to UART Bridge (COM1) opened at 115200 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
22:27:25,869 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(C:\Users\Marc\JMRI\Mon_Réseau_JMRI.jmri\roster.xml) [main]
22:27:26,169 jmri.util.FileUtilSupport             INFO  - File path program: is F:\Program Files (x86)\JMRI\ [main]
22:27:26,169 jmri.util.FileUtilSupport             INFO  - File path preference: is C:\Users\Marc\JMRI\Mon_Réseau_JMRI.jmri\ [main]
22:27:26,170 jmri.util.FileUtilSupport             INFO  - File path profile: is C:\Users\Marc\JMRI\Mon_Réseau_JMRI.jmri\ [main]
22:27:26,170 jmri.util.FileUtilSupport             INFO  - File path settings: is C:\Users\Marc\JMRI\ [main]
22:27:26,171 jmri.util.FileUtilSupport             INFO  - File path home: is C:\Users\Marc\ [main]
22:27:26,171 jmri.util.FileUtilSupport             INFO  - File path scripts: is F:\Program Files (x86)\JMRI\jython\ [main]
22:27:26,480 t.permission.DefaultPermissionManager INFO  - Permission file: C:\Users\Marc\JMRI\.permissions.xml [main]
22:27:26,484 t.permission.DefaultPermissionManager INFO  - Permission system is enabled: no [main]
23:00:57,003 conet.uhlenbrock.UhlenbrockPacketizer WARN  - Ignore LocoNet packet with bad checksum: B4 6D 01 07 [LocoNet Uhlenbrock receive handler]
23:00:57,003 conet.uhlenbrock.UhlenbrockPacketizer WARN  - run: unexpected LocoNetMessageException:  [LocoNet Uhlenbrock receive handler]
jmri.jmrix.loconet.LocoNetMessageException: null
	at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:194) [jmri.jar:?]
	at java.lang.Thread.run(Thread.java:842) [?:?]

This is what shows the System Console after trying to read all the CVs of the Basic tab (programming track, direct byte) : 

23:18:47,670 conet.uhlenbrock.UhlenbrockPacketizer WARN  - Ignore LocoNet packet with bad checksum: B4 6D 01 07 [LocoNet Uhlenbrock receive handler]
23:18:47,670 conet.uhlenbrock.UhlenbrockPacketizer WARN  - run: unexpected LocoNetMessageException:  [LocoNet Uhlenbrock receive handler]
jmri.jmrix.loconet.LocoNetMessageException: null
	at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:194) [jmri.jar:?]
	at java.lang.Thread.run(Thread.java:842) [?:?]

Hope this helps to solve the problem. Do not hesitate to request other tests if necessary.

Best regards.

bobjacobsen, 2025-03-31:
Thank you for the information!  That was very helpful.

It looks like the problem is two fold:

1) The 2Neo is sending a bad message.  A LocoNet message `B4 6D 01 07` really does have a bad checksum.  The last byte should be `27` not `07`.
2) JMRI ignores that, which gets it out of sync on the LocoNet communications.  

I'm not sure how to solve that.  Ignoring the message is the right thing to do formally:  The LocoNet specification requires it, and usually bad messages really are bad.  But this one seems to be necessary to process for the communications to not hang up.

Let me do some checking and see what I can do.  More info to follow in a bit.

bobjacobsen, 2025-03-31:
Could you try one of the JMRI installers from:

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/220/

I'm not sure that the fix in this will completely work, but I think it's worth a try.  Thank you!

Petit-Barde, 2025-03-31:
Hello and thank you for trying to solve this issue!

I installed build JMRI.5.11.4plus+jenkins+20250331T1250Z+R72ba587bcf.

Here is the System Console output while trying to load the CVs of the Basic tab : 
16:17:09,386 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
16:17:09,406 apps.util.Log4JUtil                   INFO  - This log is stored in file: C:\Users\Marc\JMRI\log\session.log [main]
16:17:09,406 apps.util.Log4JUtil                   INFO  - This log is appended to file: C:\Users\Marc\JMRI\log\messages.log [main]
16:17:09,426 apps.AppsBase                         INFO  - DecoderPro version 5.11.4plus+jenkins+20250331T1250Z+R72ba587bcf starts under Java 17.0.14 on Windows 10 amd64 v10.0 at Mon Mar 31 16:17:09 CEST 2025 [main]
16:17:09,600 apps.gui3.Apps3                       INFO  - Starting with profile Mon_Réseau_JMRI.3e1c065a [main]
16:17:09,922 jmri.util.node.NodeIdentity           INFO  - Using d1481c4e-a1a5-4c77-a97c-6868d2e2b8eb as the JMRI storage identity for profile id 3e1c065a [AWT-EventQueue-0]
16:17:10,016 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]
16:17:10,126 .loconet.uhlenbrock.UhlenbrockAdapter INFO  - Connecting Uhlenbrock via COM1 CP2102 USB to UART Bridge Controller [main]
16:17:10,136 .loconet.locobuffer.LocoBufferAdapter INFO  - LocoNet: Port Silicon Labs CP210x USB to UART Bridge (COM1) opened at 115200 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
16:17:10,634 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(C:\Users\Marc\JMRI\Mon_Réseau_JMRI.jmri\roster.xml) [main]
16:17:10,967 jmri.util.FileUtilSupport             INFO  - File path program: is F:\Program Files (x86)\JMRI\ [main]
16:17:10,968 jmri.util.FileUtilSupport             INFO  - File path preference: is C:\Users\Marc\JMRI\Mon_Réseau_JMRI.jmri\ [main]
16:17:10,968 jmri.util.FileUtilSupport             INFO  - File path profile: is C:\Users\Marc\JMRI\Mon_Réseau_JMRI.jmri\ [main]
16:17:10,968 jmri.util.FileUtilSupport             INFO  - File path settings: is C:\Users\Marc\JMRI\ [main]
16:17:10,969 jmri.util.FileUtilSupport             INFO  - File path home: is C:\Users\Marc\ [main]
16:17:10,969 jmri.util.FileUtilSupport             INFO  - File path scripts: is F:\Program Files (x86)\JMRI\jython\ [main]
16:17:11,308 t.permission.DefaultPermissionManager INFO  - Permission file: C:\Users\Marc\JMRI\.permissions.xml [main]
16:17:11,314 t.permission.DefaultPermissionManager INFO  - Permission system is enabled: no [main]
16:23:33,419 conet.uhlenbrock.UhlenbrockPacketizer WARN  - LocoNet message with opCode: b4 ended early. Expected length: 4 seen length: 2 unexpected byte: b4 [LocoNet Uhlenbrock receive handler]

The symptoms are the same as before : 
 - delays while pushing the On/Off button,
 - many 306 errors, delays and CV data in the wrong fields while trying to read the Basic tab CVs.

Trying again to read the Basic tab CV give this System Console output : 
16:31:18,494 conet.uhlenbrock.UhlenbrockPacketizer WARN  - LocoNet message with opCode: b4 ended early. Expected length: 4 seen length: 2 unexpected byte: b4 [LocoNet Uhlenbrock receive handler]
16:32:38,574 conet.uhlenbrock.UhlenbrockPacketizer WARN  - LocoNet message with opCode: b4 ended early. Expected length: 4 seen length: 2 unexpected byte: b4 [LocoNet Uhlenbrock receive handler]

Hope this help!
Best regards.

Petit-Barde, 2025-04-03:
Hello, 
the build JMRI.4.21.4 works flawlessly with the IB 2Neo : 

I installed the build JMRI.4.21.4+Rbe2a13592, and it works perfectly with the Intellibox 2Neo with an USB connection (no lag at all, the IB 2Neo reacts at once, could read all the CVs data of the speed table without any problem) : 
   - DecoderPro throttle : OK (tested on loco with ESU Locksound V5 micro DCC),
   - decoder programming (a ESU Locksound V5 micro DCC) on programming track : 
      - read : OK (tested on the speed table)
      - write : OK (tested on a single CV value)
   - accessories : not tested,
   - feddback : not tested.

Best regards,
Marc