JMRI/JMRI #13993: Decoderwerk Decoderfile LokComander

DieterFlunkert, 2025-03-30:
Decoderfile for German Lokdecoderwerk LokComander 7090x

[Decoderwerk_LokCommander.xml.zip](https://github.com/user-attachments/files/19527271/Decoderwerk_LokCommander.xml.zip)

DieterFlunkert, 2025-03-30:
This is  for the Decoderwerk LocCommander Decoder. Testet on my ESU Test Equipment.

bobjacobsen, 2025-03-30:
Did you close this on purpose?  Happy to test and include this decoder definition if you'd like to add it to JMRI.

DieterFlunkert, 2025-03-30:
I did it accidentally. It would be nice, if you could test it and if it is ok to add it to JMRI. Thanks

bobjacobsen, 2025-03-30:
It's now in PR #13994. That'll get reviewed, tested and merged, then should be in the next test release.  Thank you for providing it!

closed 2025-03-31 by danielb987