JMRI/JMRI #14049: MemoryIcons on Web Server in Panels mode

Reay-Grant, 2025-04-21:
Issue Type: __Bug 
I'm running JMRI 5.10 with Java 17. Thanks! It's great. However, I'm having strange issues with memoryicons on remote web panels. I've tried a couple of different browsers including Chrome, Firefox and Bing. 
When I change a MemoryIcon it updates on the main screen as expected. If I use the Web Server in "Enable Frames" mode, it also updates on a web browser.
However if I set the Web Server to "Use panels instead" I see no MemoryIcon and I get an error "ERROR: image not defined for memoryicon widget-3, TOstate=null, iconstate=null [] (iconnull)" in the browsers console.
If I directly type "localhost:12080/panel" into the browser I see the MemoryIcon. However if I then click on a Turnout, the browser goes to "localhost:12080/panel/Panel/Panel", the MemoryIcon disappears, and I get the error message.
My icons are stored in the preferences directory.
Any ideas?
Thanks, Reay

mstevetodd, 2025-04-21:
Hi Reay,
Please attach your panel file and any custom icons (and anything else) needed to recreate the problem.
--SteveT

Reay-Grant, 2025-04-21:
Here you go, I made a simpler test layout to try and get to the bottom of it

Here's some icons, the path I'm using is "C:\Users\reay\JMRI\My_JMRI_Railroad.jmri\resources\Icons\Reays Icons\Block Icons"
[image]
[image]
[image]
[image]

and my simpler layout is

<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/panelfile-5-5-5.xsl" type="text/xsl"?>
<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-5-5-5.xsd">
  <jmriversion>
    <major>5</major>
    <minor>10</minor>
    <test>0</test>
    <modifier />
  </jmriversion>
  <turnouts class="jmri.jmrix.internal.configurexml.InternalTurnoutManagerXml">
    <operations automate="false">
      <operation name="NoFeedback" class="jmri.configurexml.turnoutoperations.NoFeedbackTurnoutOperationXml" interval="300" maxtries="2" />
      <operation name="Raw" class="jmri.configurexml.turnoutoperations.RawTurnoutOperationXml" interval="300" maxtries="1" />
      <operation name="Sensor" class="jmri.configurexml.turnoutoperations.SensorTurnoutOperationXml" interval="300" maxtries="3" />
    </operations>
    <defaultclosedspeed>Normal</defaultclosedspeed>
    <defaultthrownspeed>Restricted</defaultthrownspeed>
    <turnout feedback="DIRECT" inverted="false" automate="Off">
      <systemName>IT1</systemName>
    </turnout>
  </turnouts>
  <memories class="jmri.managers.configurexml.DefaultMemoryManagerXml">
    <memory value="10:36 pm">
      <systemName>IMCURRENTTIME</systemName>
    </memory>
    <memory value="-">
      <systemName>IMIBICON24</systemName>
      <userName>Block 24 Icon</userName>
    </memory>
    <memory value="1.0">
      <systemName>IMRATEFACTOR</systemName>
    </memory>
  </memories>
  <signalmastlogics class="jmri.managers.configurexml.DefaultSignalMastLogicManagerXml">
    <logicDelay>500</logicDelay>
  </signalmastlogics>
  <paneleditor class="jmri.jmrit.display.panelEditor.configurexml.PanelEditorXml" name="Panel" x="772" y="505" height="380" width="435" editable="yes" positionable="yes" showtooltips="yes" controlling="yes" hide="yes" panelmenu="yes" scrollable="both" redBackground="255" greenBackground="255" blueBackground="255">
    <turnouticon turnout="IT1" x="23" y="28" level="7" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" tristate="false" momentary="false" directControl="false" class="jmri.jmrit.display.configurexml.TurnoutIconXml">
      <icons>
        <closed url="program:resources/icons/smallschematics/tracksegments/os-righthand-west-closed.gif" degrees="0" scale="1.0">
          <rotation>0</rotation>
        </closed>
        <thrown url="program:resources/icons/smallschematics/tracksegments/os-righthand-west-thrown.gif" degrees="0" scale="1.0">
          <rotation>0</rotation>
        </thrown>
        <unknown url="program:resources/icons/smallschematics/tracksegments/os-righthand-west-unknown.gif" degrees="0" scale="1.0">
          <rotation>0</rotation>
        </unknown>
        <inconsistent url="program:resources/icons/smallschematics/tracksegments/os-righthand-west-error.gif" degrees="0" scale="1.0">
          <rotation>0</rotation>
        </inconsistent>
      </icons>
      <iconmaps />
    </turnouticon>
    <memoryicon memory="IMIBICON24" x="101" y="100" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="false" editable="true" size="15" style="0" red="0" green="0" blue="0" hasBackground="no" fixedWidth="73" fixedHeight="15" justification="left" selectable="no" updateBlockValue="yes" class="jmri.jmrit.display.configurexml.MemoryIconXml" defaulticon="preference:resources/Icons/Reays Icons/Block Icons/BLANK-L.gif">
      <memorystate value="12L" icon="preference:resources/icons/Reays Icons/Block Icons/Blue HST-L.gif" />
      <memorystate value="-" icon="preference:resources/icons/Reays Icons/Block Icons/BLANK-L.gif" />
      <memorystate value="12R" icon="preference:resources/icons/Reays Icons/Block Icons/Blue HST-R.gif" />
      <memorystate value="?" icon="preference:resources/icons/Reays Icons/Block Icons/UNKNOWN-L.gif" />
    </memoryicon>
  </paneleditor>
  <filehistory>
    <operation>
      <type>app</type>
      <date>Mon Apr 21 15:26:57 BST 2025</date>
      <filename>PanelPro (v5.10.0)</filename>
    </operation>
    <operation>
      <type>Load OK</type>
      <date>Mon Apr 21 15:26:59 BST 2025</date>
      <filename>/C:/Users/reay/JMRI/My_JMRI_Railroad.jmri/My Layout.xml</filename>
      <filehistory>
        <operation>
          <type>app</type>
          <date>Mon Apr 21 14:41:38 BST 2025</date>
          <filename>PanelPro (v5.10.0)</filename>
        </operation>
        <operation>
          <type>Store</type>
          <date>Mon Apr 21 14:44:05 BST 2025</date>
          <filename>C:\Users\reay\JMRI\My_JMRI_Railroad.jmri\My Layout.xml</filename>
        </operation>
      </filehistory>
    </operation>
    <operation>
      <type>Store</type>
      <date>Mon Apr 21 15:56:46 BST 2025</date>
      <filename>C:\Users\reay\JMRI\My_JMRI_Railroad.jmri\My Layout prog size.xml</filename>
    </operation>
  </filehistory>
  
</layout-config>

Sorry it won't let me upload an xml file
Reay

Reay-Grant, 2025-04-21:
Ah that's not going to help

[My Layout.txt](https://github.com/user-attachments/files/19837716/My.Layout.txt)

there you go, in text file
Cheers

Reay-Grant, 2025-04-21:
> Thanks, I can load that fine, but I'm obviously missing something. All I
> see on the JMRI panel is a turnout icon (see screenshot). I don't see your
> custom files at all. Are you taking some additional action to make them
> show up? If I click on the turnout icon, it changes to the Closed or Thrown
> image as expected.
>
> image.png (view on web)

Yeah it's literally one turnout and one MemoryIcon. If you open the
MemoryIcon table it should be there.
Hope that helps
Reay

mstevetodd, 2025-04-21:
It looks like your MemoryIcon is expecting values like "12L", "12R", "-", "?". I can manually enter one of those in the Memory table, but surely you're not doing that? Are you also running a script or something to set those values?

Regardless, once I put manually "12L" as the memory value, I can see the Memoryicon in the JMRI panel, and can duplicate the error in the browser console. 
My immediate guess is that one (or more) of the spaces in the path and filename are causing the error. I will look into a fix, but my suggestion would be "don't do that". I avoid spaces and several other special characters (anything except "-" and "_" basically) as browsers need to escape them, so the programmer has to undo that, etc.

--SteveT

Reay-Grant, 2025-04-21:
Thanks for the prompt response! Yeah that makes sense. I normally run it
off a script but I stripped it all down to make sure nothing else is
interfering with JMRI (I'm originally an ASM programmer from the days when
you wrote your own debugger). So a space is getting translated into %20
then...
I'll try that at my end.
Cheers Reay

mstevetodd, 2025-04-21:
Ok, thanks for trying that. I was hoping it was that simple.

The page "/panel" is actually a list of available frames shown as "thumbnails", just like the home page. Thise images are from the "/frame" server.

I'll debug the panel.js code when I get a chance. I've never seen MemoryIcons with names like that, I'm used to seeing "0", "1", "2", etc. Possible that assumption made it into the code. I'll have to start by figuring out how you even got those values in there. :)

Until then, you'll need to use the /frame URL, as it sends a "screenshot" of the JMRI panel so it always matches.

mstevetodd, 2025-04-21:
Reay,
I tried to edit the MemoryIcon in JMRI, and failed. How (exactly) did you assign those values to those icons? I rarely use Panel Editor, so I may be missing something obvious.
--SteveT

Reay-Grant, 2025-04-21:
You have to do it in the My Layout.xml file directly
Regards Reay

mstevetodd, 2025-04-21:
Oh.

In that case, include a value for "null", such as 
`      <memorystate value="null" icon="preference:resources/icons/Reays Icons/Block Icons/UNKNOWN-L.gif" />
`
That will get past the error on the initial icon drawing.

Background: Each widget is initially drawn using (only) the data in the XML, then the current state is requested from JMRI. Subsequent changes to state are sent automatically. Each widget is redrawn as updates arrive from the server. Only the initial draw was failing.

Reay-Grant, 2025-04-21:
Hmm that gives me a placeholder image, when I click on it I get a 404 error

mstevetodd, 2025-04-21:
Didn't you changed your path earlier?
Use the exact same syntax as the other values, just set the value to "null". 
```
    <memoryicon memory="IMIBICON24" x="101" y="100" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="false" editable="true" size="15" style="0" red="0" green="0" blue="0" hasBackground="no" fixedWidth="73" fixedHeight="15" justification="left" selectable="no" updateBlockValue="yes" class="jmri.jmrit.display.configurexml.MemoryIconXml" defaulticon="preference:resources/Icons/Reays Icons/Block Icons/UNKNOWN-L.gif">
      <memorystate value="12L" icon="preference:resources/icons/Reays Icons/Block Icons/Blue HST-L.gif" />
      <memorystate value="-" icon="preference:resources/icons/Reays Icons/Block Icons/BLANK-L.gif" />
      <memorystate value="12R" icon="preference:resources/icons/Reays Icons/Block Icons/Blue HST-R.gif" />
      <memorystate value="?" icon="preference:resources/icons/Reays Icons/Block Icons/UNKNOWN-L.gif" />
      <memorystate value="null" icon="preference:resources/icons/Reays Icons/Block Icons/UNKNOWN-L.gif" />
    </memoryicon>

```

mstevetodd, 2025-04-21:
Please attach the edited file and I'll take a look.

Reay-Grant, 2025-04-21:
> Still no, I'm afraid!
> I've sent a screenshot zoomed in on the placeholder icon, it says
> IMIBICON24 by it's top right corner.
> Regards, Reay

There's a MemoryIcon code sample at JMRI>xml>samples and that has exactly
the same issue that I'm having.
Regards, Reay

Reay-Grant, 2025-04-21:
Ok result! If I add a null value to the sample code, it then works.
I guess that sample needs updating.
Thanks very much for your help, I reckon I can probably figure it out from
here.
Regards, Reay

mstevetodd, 2025-04-21:
The null value is a workaround, not a good fix. Please leave this issue open and I'll work on a proper fix (sometime). 
I will also find and update the sample as needed to work properly with the resulting panel.js code.

Reay-Grant, 2025-04-21:
Update: my user folder is called reay, but Windows replaces it with my full
name (Reay Grant) in the path to the preferences folders. If I copy the
icons and their subfolders to the program: resources folder it works fine!
I'll let you know if I figure out anything else.
Regards, Reay

dsand47, 2025-04-21:
> You have to do it in the My Layout.xml file directly
> Regards Reay
> […](#)

How did you discover the "memorystate" feature?  I am not aware of any documentation describing it.  I also don't find anything about using it.

Reay-Grant, 2025-04-22:
It's in the sample code at Program Files(x86)>JMRI>xml>samples
Regards, Reay

mstevetodd, 2025-04-22:
The fix turned out to be fairly simple. I've created PR 14052 for it. https://github.com/JMRI/JMRI/pull/14052

This fix does NOT require adding the "null" workaround, and works with the MemoryIconExample.xml from the sample folder as well as the one provided by @Reay-Grant in this issue.

@Reay-Grant If you'd like to try it, replace your web/js/panel.js file with this one: https://github.com/JMRI/JMRI/blob/259e28db3ab0a61b465e50ced7bd52ec1cdb3c0b/web/js/panel.js

Reay-Grant, 2025-04-22:
Yep that's done it. Get rid of all the spaces in the file names and paths,
add a "null" value, and move the icons to the program:resources directory.
Yep it's not particularly well documented but we got there in the end!
Thanks for all your help again. If I figure out any more details, I'll let
you know.
Regards, Reay

mstevetodd, 2025-04-22:
The fix turned out to be fairly simple. I've created PR 14052 for it. https://github.com/JMRI/JMRI/pull/14052

This fix does NOT require adding the "null" workaround or removing the spaces, and works with the MemoryIconExample.xml from the sample folder as well as the one provided by @Reay-Grant in this issue.

@Reay-Grant If you'd like to try it, replace your web/js/panel.js file with this one: https://github.com/JMRI/JMRI/blob/259e28db3ab0a61b465e50ced7bd52ec1cdb3c0b/web/js/panel.js

Reay-Grant, 2025-04-22:
It's working fine, as long as I use the icons in the program: resources
directory.
It's very odd, the path is identical from: resources onwards, and I've
copied the entire folder, so it should be the same as far as software is
concerned.
The only thing I can see, is Microsoft replacing the name of my user folder
("reay") with my Microsoft account name ("Reay Grant").
I'll keep you posted, although I'm off to bed now.
Thanks yet again
Reay

commit 2aac6b69 (2025-04-23): Merge pull request #14052 from mstevetodd/master

panel.js - fix to memorystate memoryicon, closes #14049
closed 2025-04-23 by danielb987