JMRI/JMRI #14050: Reading CVs from Zimo MXULF

jonasj-245, 2025-04-21:
Hi

I like to get some help when JMRI doesn't provide the right result from the decoder.

Equipment seems to work, as it is reading out the CV, but the problem is that it comes in the random order, for example it returns different numbers even with the same CV input.

From PC Seq 163 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 125 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 163 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 126 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 163 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 127 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 163 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 128 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 164 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From CS Seq 129 (S) Reply 2 to/from CS (MX1) Read CV Loco: 0 DCC CV: 29 Set To: 14
From PC Seq 165 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 130 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 165 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 131 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 165 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 132 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 165 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 133 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 166 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From CS Seq 134 (S) Reply 2 to/from CS (MX1) Read CV Loco: 0 DCC CV: 17 Set To: 192
From PC Seq 167 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 135 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 167 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 136 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 167 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 137 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 167 (S) Prim to/from CS (MX1) Read CV Loco: 0 DCC CV: 1
From CS Seq 138 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From PC Seq 168 (S) Ack/Reply 1 to/from CS (MX1) Prog CV 
From CS Seq 139 (S) Reply 2 to/from CS (MX1) Read CV Loco: 0 DCC CV: 18 Set To: 128

JMRI version 5.11.5+R6be2a2b4e8
Java version 17.0.12
MXULF 00.84.112 (applications ZSP and ZCS work)
Windows 11

jonasj-245, 2025-04-21:
Hi

I am using the Single CV Programmer and it giving me different results on the same CV1

https://groups.io/g/jmriusers/topic/reading_cvs/112373569

christopherjhharvey, 2025-06-06:
All CV programming works fine for me when the MXULFA is running 0.84.58 in both JMRI (JMRI version 5.10 on Java 11 and Windows 11) and using ZCS (4.21.040) for all loco's I own using exactly the same setup.

If I upgrade to MXULFA FW version 0.84.93 programming is no longer reliable in JMRI with the symptoms as described.

Reading the release notes here: https://www.zimo.at/web2010/support/UpdateMXULF_EN.htm  I would initially suspect the USB interface changes released in FW 0.84.92:

> USB interface:
>     Correction in the DCC_ON protocol (ZSP control panel): MXULF no longer inserts its own checksum when sending DCC packets from the PC.
>     New commands for handling the various USB VCOM protocols.
>     Correction in the binary protocol: Query x10/x11 (see documentation) is now answered with correct rail voltage '0' for rail off. Relevant for ZCS.

jonasj-245, 2025-06-07:
How should I interpret the fact that we are not supposed to use a higher FW version that supports MN/MS decoders?

uli-2016, 2025-06-18:
The current firmware version of the MXULFA is not correctly supported by JMRI. Adjustments to the communication between JMRI and ZIMO MXULFA are necessary here. Also the possibility of PoM with MXULFA should be added. I think this is what this issue is for.

uli-2016, 2026-02-02:
Today, I made log recordings with versions 4.20 and 5.11.7 (Java 17.0.15). I first tried to identify the locomotive and then read the type from the decoder. Everything was correct at 4.20.

[monitorLoG_DeoderPro4.20.log](https://github.com/user-attachments/files/25024272/monitorLoG_DeoderPro4.20.log)

[monitorLogDecoderPro5.11.7.log](https://github.com/user-attachments/files/25024300/monitorLogDecoderPro5.11.7.log)

In version 5.11, values are returned, but they are assigned incorrectly. For example, I saw the value CV7=35 as the value for the manufacturer.

I am happy to carry out further tests.

uli-2016, 2026-02-11:
Perhaps this will help solve the problem:
https://github.com/ZIMO-Elektronik/ULF_COM

[The serial communication of the MX1_MXULF_V5_9.pdf](https://github.com/user-attachments/files/25228722/The.serial.communication.of.the.MX1_MXULF_V5_9.pdf)

Note: JMRI currently only supports programming on the programming track. However, MXULF support RailCom, so main track programming (PoM) should also be possible.

rlorenz-afk, 2026-09-30:
Hello,
The problem exist:
- No POM, ESU decoder reading need Terrible long Time
- Error reading setting with Mxulf 0.85.12, only page Modus selectable.
- Redaing with Tams RedBox an Mxulf give differend settings vor CV's