JMRI/JMRI #14069: Problem with opening LCC Hub on startup

JohnCallingham, 2025-04-29:
I have created a configuration profile which contains two connections, one for LCC and one for MERG. In the Preferences->Start Up I selected to Perform Action, Start OpenLCB Hub. After saving and restarting the MERG Hub Control window opens instead of the LCC Hub Control.

I can manually open the LCC Hub Control but am not able to do so from the start up options.

bobjacobsen, 2025-04-29:
Are the connections defined in that order, with LCC as connection 1 and MERG as connection 2? What system letters have you given to the two connections?

JohnCallingham, 2025-04-29:
Connection 1 is LCC, prefix is L. Connection 2 is MERG, prefix is M.

bobjacobsen, 2025-04-30:
PR #14071 has a fix for this.  You should be able to pick that up in a development build in a couple days:   https://builds.jmri.org/jenkins/job/development/job/packages/