JMRI/JMRI #14081: Train stops when released from WiThrottle

ac6000tr, 2025-05-03:
When I go to release a train from WiThrottle on my iPhone the train stops, I believe there should be a way to control this behavior as it is not present when I use the DCC-EX WiThrottle server.

mstevetodd, 2025-05-07:
I am working with user to debug and diagnose this. 

Current findings indicate that EX-CS is setting the speed to 0 when it receives the Drop Cab command `<- 1234>` from JMRI. I'll look to verify that in the EX-CS code.

mstevetodd, 2025-05-07:
Found the code in DCC.cpp, and the Forget Cab command includes eStopping the loco. This is not a recent change. I'm going to ask that they include this behavior in the documentation.
```
void DCC::forgetLoco(int cab) {  // removes any speed reminders for this loco
  setThrottle2(cab,1); // ESTOP this loco if still on track
```
The user found that JMRI only sends the Forget if there are 0 clients for that loco address, so he is currently keeping a JMRI throttle window open.

danielb987, 2025-06-10:
A suggestion is to add an advanced setting to the DCCpp connection to not send "Forget Cab command".