JMRI/JMRI #14082: Webserver example of IDTags displayed as icons track diagram

nigelcliffe, 2025-05-03:
As discussed a few weeks ago on the JMRI Developer's list. 

The attached is a ZIP file with an example of showing ID Tags as icons, which move to different places on the webpage depending on their most recent report.   The example shows cars being switched in a simple yard, the principles shown could be used to show locomotives or train movements.  

Recommend this ZIP file goes into the JMRI distribution, in the "JMRI/Web" area.    

The ZIP contains a HTML file, a sub-folder of PNG images, and some notes on the example in a Word file (source file) and PDF.

I don't know if it merits a separate sub-folder, though I have a related example to write up which uses OperationsPro "locations", "spurs" and "cars", and in addition to showing current car location, will indicate whether a car has been moved to its destination, or is at a different location to its destination.  

[JMRI Webserver Tag Visual Location.zip](https://github.com/user-attachments/files/20024689/JMRI.Webserver.Tag.Visual.Location.zip)

bobjacobsen, 2025-05-05:
I think we should put this in a new directory in the distribution at `web/idtag_examples`.  That's parallel to the existing `web/logixng_examples` directory.

If nobody has objections, I'll make a PR to that effect this evening or early tomorrow.

bobjacobsen, 2025-05-06:
See Issue #14088

PR #14088 (2025-05-07): Added files from Issue 14082 - web idtags examples
See Issue #14082 for information. Closes #14082
closed 2025-05-07 by danielb987