JMRI/JMRI #14104: Startup problem using Marklin-CS2 (TCP) & Marklin CS2 (UDP) protocol

ghfbsd, 2025-05-12:
There are two problems with the present implementation(s) of the Marklin CS2 protocol.

1) On startup and at shutdown, JMRI sends a CS2 protocol packet that is all zeroes.  This does not cause problems, but the packet is not defined in the public protocol definition.  (Rocrail does not emit it at startup or shutdown.)

2) On startup, the system does NOT send a "CAN BOOT" (0x1b) protocol packet.  The packet resets the Gleisbox/trackbox and advises it to start passing commands to the locos & accessories on the rails; without it, there is not response to any following commands from JMRI.  In order to kick the system into responding, I have to run Rocrail briefly to activate the Gleisbox and then switch to running it with JMRI.  The "CAN BOOT" packet is not fully documented in the public protocol definition, but Rocrail uses it to start operations.  (I can provide a specimen CS2 packet if you cannot find more information from other online sources.)

The hardware used is a Marklin 60116 trackbox attached through home-brew wifi-based controller (see [this link](https://github.com/ghfbsd/pico_rocrail_can_tcp_gateway)), which has been thoroughly tested with the Rocrail software.   

System info:
MacOS 11.6.5 on MacBook Air (M1, 2020)

JMRI release info:
Marklin-CS22: using CS2 via TCP network on ...
JMRI version 5.11.5+R6be2a2b4e8
Java version 11.0.27 (en)

ralflang, 2025-11-21:
@ghfbsd 

This does not create problems using the CAN-Digital-Bahn CC-Schnitte or CC-Schnitte 2.0
Sending CAN-BOOT should be optional as blocking the Gleisbox without a MS2 controller attached is a security feature.

Can you describe what kind of problems you see? The current state as you describe looks like "works as intended".

Using JMRI to impersonate a MS2 or CS2 (and actually send a ping with a high score serial to take over master role) is something I intend to implement but life hasn't been kind to my open source endeavours lately and I keep getting derailed so don't hold your breath.

ralflang, 2025-11-21:
Can you send of full bitstring capture of the CAN Boot command sent by an MS2 with recent (3.11 or 4.x) firmware and, if possible, capture one from two different MS2 to understand if it contains the device serial or is a constant string?

I think I can do this in two evolutions / PRs:

- Implement the actual command and a trigger in CdB/CS2 specific menu.
- If it works as intended, add a flag in software config to send a CAN-BOOT automatically on startup.

However, sending a CAN-BOOT on software startup may be undesirable if traffic is running already so a more robust implementation would be

- on boot, watch the can bus for at least the ping interval
- only if no traffic is observed, send the CAN-Boot.

What do you think?

Found some attempts to document CAN-Boot (German) https://www.stummiforum.de/t122854f7-M-rklin-CAN-Protokoll-x-B-commands-updates.html

ghfbsd, 2025-11-21:
Here is a packet trace with Rocrail driving the Märklin CS2 emulation device.  The device was started and in an idle state waiting for CS2 packets over TCP.  Then Rocrail was started, and began communicating with the emulated CS2.

Note that the first CS2 command sent is a CAN BOOT command.

```
Feedback: 8 channels, "S88" module 1.
TCP <-> CAN packet hub (AY285)
Available at 10.0.1.31 as CS2hub-2af98b, port 15731.
CAN Joy-IT board initialized successfully, waiting for traffic.
TCP connection made, waiting for traffic.
TCP -> CAN 0036 e32a 00 0000 0000 0000 0000
    C CAN BOOT 1B (everybody)
TCP -> CAN 0030 e32a 00 0000 0000 0000 0000
    C PING 18 (everybody)
CAN -> TCP 0037 3941 08 4d56 7417 0401 0034
    R CAN BOOT 1B (4d56741704010034): 
CAN -> TCP 0031 4f20 08 4746 e70b 013e 0010
    R PING 18 (4746e70b): Gleisbox 60116 ver 1.62
TCP -> CAN 003a e32a 05 4746 e70b 0000 0000
    C 1d hash e32a prio: 0 data: 47 46 e7 0b 00
CAN -> TCP 003b 0301 08 0300 0000 0004 b50b
    R 1d hash 0301 prio: 0 data: 03 00 00 00 00 04 b5 0b
CAN -> TCP 003b 0302 08 3630 3131 3600 0000
    R 1d hash 0302 prio: 0 data: 36 30 31 31 36 00 00 00
CAN -> TCP 003b 0303 08 476c 6569 7362 6f78
    R 1d hash 0303 prio: 0 data: 47 6c 65 69 73 62 6f 78
CAN -> TCP 003b 0304 08 2032 0000 0000 0000
    R 1d hash 0304 prio: 0 data: 20 32 00 00 00 00 00 00
CAN -> TCP 003b 4f20 05 4746 e70b 0000 0000
    R 1d hash 4f20 prio: 0 data: 47 46 e7 0b 00
TCP -> CAN 003a e32a 05 4746 e70b 0100 0000
    C 1d hash e32a prio: 0 data: 47 46 e7 0b 01
CAN -> TCP 003b 0301 08 01fd 30f0 e0c0 000f
    R 1d hash 0301 prio: 0 data: 01 fd 30 f0 e0 c0 00 0f
CAN -> TCP 003b 0302 08 0670 06c2 0767 080c
    R 1d hash 0302 prio: 0 data: 06 70 06 c2 07 67 08 0c
CAN -> TCP 003b 0303 08 5452 4143 4b00 302e
    R 1d hash 0303 prio: 0 data: 54 52 41 43 4b 00 30 2e
CAN -> TCP 003b 0304 08 3030 0032 2e35 3000
    R 1d hash 0304 prio: 0 data: 30 30 00 32 2e 35 30 00
CAN -> TCP 003b 4f20 05 4746 e70b 0100 0000
    R 1d hash 4f20 prio: 0 data: 47 46 e7 0b 01
TCP -> CAN 003a e32a 05 4746 e70b 0200 0000
    C 1d hash e32a prio: 0 data: 47 46 e7 0b 02
CAN -> TCP 003b 0301 08 03fd c00c 30c0 0000
    R 1d hash 0301 prio: 0 data: 03 fd c0 0c 30 c0 00 00
CAN -> TCP 003b 0302 08 039d 04b2 0a1e 0c49
    R 1d hash 0302 prio: 0 data: 03 9d 04 b2 0a 1e 0c 49
CAN -> TCP 003b 0303 08 564f 4c54 0031 302e
    R 1d hash 0303 prio: 0 data: 56 4f 4c 54 00 31 30 2e
CAN -> TCP 003b 0304 08 3030 0032 372e 3030
    R 1d hash 0304 prio: 0 data: 30 30 00 32 37 2e 30 30
CAN -> TCP 003b 4f20 05 4746 e70b 0200 0000
    R 1d hash 4f20 prio: 0 data: 47 46 e7 0b 02
TCP -> CAN 003a e32a 05 4746 e70b 0300 0000
    C 1d hash e32a prio: 0 data: 47 46 e7 0b 03
CAN -> TCP 003b 0301 08 0400 0c08 f0c0 0000
    R 1d hash 0301 prio: 0 data: 04 00 0c 08 f0 c0 00 00
CAN -> TCP 003b 0302 08 006b 00a4 00cd 00db
    R 1d hash 0302 prio: 0 data: 00 6b 00 a4 00 cd 00 db
CAN -> TCP 003b 0303 08 5445 4d50 0030 2e30
    R 1d hash 0303 prio: 0 data: 54 45 4d 50 00 30 2e 30
CAN -> TCP 003b 0304 08 0038 302e 3000 4300
    R 1d hash 0304 prio: 0 data: 00 38 30 2e 30 00 43 00
CAN -> TCP 003b 4f20 05 4746 e70b 0300 0000
    R 1d hash 4f20 prio: 0 data: 47 46 e7 0b 03
TCP -> CAN 003a e32a 05 4746 e70b 0400 0000
    C 1d hash e32a prio: 0 data: 47 46 e7 0b 04
CAN -> TCP 003b 4f20 05 4746 e70b 0400 0000
    R 1d hash 4f20 prio: 0 data: 47 46 e7 0b 04
[... 39 lines cut ...]
    C SYSTEM STATUS 00/0B (4746e70b): channel 4
CAN -> TCP 0001 4f20 08 4746 e70b 0b01 000f
    R SYSTEM STATUS 00/0B (channel 1) TRACK 0.00A
CAN -> TCP 0001 4f20 08 4746 e70b 0b04 0038
    R SYSTEM STATUS 00/0B (channel 4) TEMP 20.46C
TCP -> CAN 0000 e32a 08 0000 0000 2013 283c
    C SYSTEM CLOCK 00/20 (00000000): 19:40 tick 60s
TCP -> CAN 0000 e32a 06 4746 e70b 0b01 0000
    C SYSTEM STATUS 00/0B (4746e70b): channel 1
TCP -> CAN 0000 e32a 06 4746 e70b 0b04 0000
    C SYSTEM STATUS 00/0B (4746e70b): channel 4
CAN -> TCP 0001 4f20 08 4746 e70b 0b01 000f
    R SYSTEM STATUS 00/0B (channel 1) TRACK 0.00A
CAN -> TCP 0001 4f20 08 4746 e70b 0b04 0038
    R SYSTEM STATUS 00/0B (channel 4) TEMP 20.46C
TCP -> CAN 0000 e32a 06 4746 e70b 0b01 0000
    C SYSTEM STATUS 00/0B (4746e70b): channel 1
TCP -> CAN 0000 e32a 06 4746 e70b 0b04 0000
    C SYSTEM STATUS 00/0B (4746e70b): channel 4
CAN -> TCP 0001 4f20 08 4746 e70b 0b01 000f
    R SYSTEM STATUS 00/0B (channel 1) TRACK 0.00A
CAN -> TCP 0001 4f20 08 4746 e70b 0b04 0038
    R SYSTEM STATUS 00/0B (channel 4) TEMP 20.46C
...
```

bobjacobsen, 2026-02-16:
@ralflang Do you have time to look at this?  It's well beyond the parts of the code that I understand, unfortunately.

ralflang, 2026-02-19:
It looks like we turned 0x1B instead of 0xB1. I am trying to figure out.

ralflang, 2026-02-19:
@ghfbsd I have provided a PR with a potential fix. Unfortunately I am still on a business trip and will take some time till I am back at a physical layout. Do you have any means to test this from source code or do you need a snapshot build (JAR) delivered?

ghfbsd, 2026-02-20:
@danielb987 - thanks for the Jenkins pointer.  I obtained a test .dmg installer and tested the fix.

@ralflang - I don't see any faulty SYSTEM COMMANDs sent by JMRI any more.  However, the CAN BOOT command to activate the gleisbox isn't quite right (from the Panel Pro `Marklin-CS2` -> `Send MCAN BOOT` tab).  In order to kick the gleisbox into paying attention when there is not a CS2 or an MS2 attached, you have to give it a CAN BOOT command with DLC = 5 and data byte 4 = 0x11.  See this Swift code snippet that generates the packet:

```
   public static func boot() -> MarklinCANMessage {
        let message = MarklinCANMessage(prio: 0,        // The magic Gleisbox start command,                            
                                        command: 0x1b,  // courtesy of https://github.com/GBert/railroad/can2udp        
                                        resp: 0,
                                        hash: hash,
                                        dlc: 5,
                                        byte0: 0,
                                        byte1: 0,
                                        byte2: 0,
                                        byte3: 0,
                                        byte4: 0x11,
                                        byte5: 0,
                                        byte6: 0,
                                        byte7: 0
        )
        return message 
    }
```
Sending that packet would be the appropriate thing to be under that menu tab.

Alternatively (and more user-friendly), if you make JMRI do a PING when it connects to a Marklin protocol device and listen to the responses, you can  determine whether there is one (or more) MS2s, a CS2/CS3 or just a gleisbox running the layout.

For gleisbox-only devices that JMRI can run a layout through, see [Stummi forum's Gleisbox als Zentrale](https://www.stummiforum.de/t160368f7-Gleisbox-als-Zentrale.html), [locoduino's gleisbox centrale](https://www.locoduino.org/spip.php?article361), [wifi-hub](https://github.com/ghfbsd/pico_rocrail_can_tcp_gateway), among others.

closed 2026-02-22 by danielb987