JMRI/JMRI #14105: Update a Startup script to ignore LCC Turnouts and Sensors

Cliff-Anderson, 2025-05-12:
The Jython script named "**InitAllTurnoutsSensors.py**" as currently included in the JMRI distributed downloads treats LCC controlled Turnouts and Sensors as if they were to behave in similar way to conventional DCC controlled Turnouts and Sensors that can be in UNKNOWN states at Startup.

In fact the nodes controlling the LCC Turnouts and Sensors remember each of their last known controlling EventIDs at shutdown and either repeat those EventIDs or else somehow restore each item to the same state when powered up.  (At least that is how I understand _event driven_ vs _state driven_ startup.)

If the existing script is run after the powerup, the states of the LCC Turnouts and LCC Sensors will be overridden to either CLOSED or INACTIVE respectively.  There is no currently described method to force a global update on the LCC network that could be activated within a script.  

The modified version supplied here now ignores all LCC Turnouts and Sensors and treats them as if they were Internal Turnouts and Sensors.  This allows a layout that has been using DCC defined Turnouts and/or DCC defined Sensors and is now being expanded to include LCC controlled equipment to cope with both styles of equipment.  

[InitAllTurnoutsSensors.zip](https://github.com/user-attachments/files/20173790/InitAllTurnoutsSensors.zip)

danielb987, 2025-05-21:
Thank you. I have created PR #14120. Can you check that I have got it right?

danielb987, 2025-05-21:
@Cliff-Anderson 
There is some discussion about your changes in PR https://github.com/JMRI/JMRI/pull/14120. It would be good if you could join it.