JMRI/JMRI #14113: Can't read/write LNCV for GBM16XL

SmilestonesCH, 2025-05-16:
Hello,
I'm trying to program the LNCV of the GBM16XL feedback module from Blücher Electronique.
https://www.bluecher-elektronik.de/produkte/gleisbesetztmelder-gbm16xl-vers-1-6/
The feedback module is connected to an Intellibox I from Uhlenbrock. The PC is connected to the Intellibox via Uhlenbrock's LocoNet 63120 interface.
The documentation states that the article number to be entered for the GBM16XL is 12350.
Unfortunately no messages are exchanged when I try to read or describe an LNCV.

What's the problem?

[profile.zip](https://github.com/user-attachments/files/20242424/profile.zip)
[session.log](https://github.com/user-attachments/files/20242425/session.log)
[systemInfo.txt](https://github.com/user-attachments/files/20242426/systemInfo.txt)

klk32003, 2025-05-18:
I don't think this is an JMRI issue.

Maybe it's a wiring issue or an Intellibox version issue. My Intellibox has version 2.001 and I can enter the article number of my GBM16XN. I tried to read LNCV 8 and got 60 (but actually I use the config tool to do the configuration).
Have you tried to contact the manufacturer?

SmilestonesCH, 2025-05-19:
Hello,

The connection is ok because when I push the STOP or the GO button on the IB I see the messages in the LocoNet Monitor window of DecoderPro.
When I try to read or write a CV from the window "Manage LNCV Modules" I see absolutely no LocoNet messages in the Monitor window...
Attached my connection setup from the Preferences and the configuation in the "Manage LNCV Modules" window's.

Any advices are welcome.

Best regards
Yann Monbaron

SmilestonesCH, 2025-05-19:
> Yann,
>
> I didn't know that you use the "Manage LNCV Modules" function in DecoderPro.
> I have done the LNCV read test by using the Intellibox LocoNet Prog menu.
> Have you already tried this?
>
> IB1_LocoNet_Prog.jpg (view on web)<https://github.com/user-attachments/assets/7ef1c731-93c9-4ba4-8fe9-08eadf4beea1>
>
> So my next and first time test was with JMRI "Manage LNCV Modules".
> I had to enter the article number without the trailing 0.
> Then I could see the traffic in the monitor.
>
> Klaus

Hello Klaus,

Thank you for your answer.
Ah yes of course from the Intellibox it works perfectly. What doesn't work is from JMRI.
On your photo you have entered the article number 12340 but in the GBM16XL documentation it says that you have to enter the article number 12350. Are you using a GBM16XL?
In JMRI I've tried with 1235 and 1234 but still nothing when I try to read an LNCV... :-(

Best regards
 
Yann

klk32003, 2025-05-20:
> On your photo you have entered the article number 12340 but in the GBM16XL documentation it says that you have to enter the article number 12350. Are you using a GBM16XL?

No, I use a GBM16XN.

Here is an example of a "Manage LNCV modules" window:

[image]

Do you get the "1 Module discovered" message?

Another question: can you drive a locomotive with an JMRI throttle?