JMRI/JMRI #14123: Make Packet Analyser Output Usable in Java Code

vandoornw, 2025-05-23:
SUMMARY:
Create a programmatic interface to obtain packet analyser output.

DETAIL:
At the moment, the NCE Packet Analyser is supported by the provision of a window that shows the output, raw or formatted, and allows control via an elementary interface through the keyboard.

It would be very helpful to me if there were a programmatic interface in addition to this. It could be as beautiful and intricate as the developer interested in creating it likes to make it (filtering similar to what the existing window allows; specific locos, specific accessories, specific commands - whatever), but the most basic of interfaces would do me fine - that would be provision of a listener allowing me access to the raw text lines that already form the most basic outputs in the analyser window.

I shall use this in my own code (I use JMRI as a library to program against, as was one of the original intentions of the JMRI project, see the trailing 'I' in the name!) for several things I have in mind:
- detecting dangerous conditions caused by users, such as: going in the wrong direction, jumping signals;
- detecting tunouts moved via throttles on command stations that do not support that;
- automatically maintaining times and distances a loco has operated;
- detecting and undoing momentum changes;
- automatically operating lights if the user forgets;
... and no doubt more, eventually.

I'm totally happy doing all the required parsing in my own code, so that the istener (I'm assuming a llistener, but if it's a different type of interface - fine) can be kept extremely simple.

I do not want to make local changes to JMRI code. I have never needed to do that yet, and would like to keep it that way. My own code all resides in my own project, referencing JMRI within NetBeans for development. My code is then used at run-time as an exported jar only (and loads of config-files JMRI need not know about). I like to keep it this clean, so that I can at any time udate JMRI without having to keep in mind my own additions and modifications.

bobjacobsen, 2025-05-25:
Is there a reason to build this on the NCE Packet Analyzer window?

That window has a lot of shortcomings:  You have to start it manually, you need to have an NCE connection, it's difficult to locate it programmatically, etc.

If instead there was a script that provided access to the same information, would that meet the need for this? That could start automatically and provide straight-forward access to the information.

KenC57, 2025-05-25:
From my last rework of the tool, to support the newer analyser, I don't recall a true reason it is tied to NCE. I think it really could be independent of the NCE connection. I suspect the only connection is which menu and when it gets added to the menus.

vandoornw, 2025-05-25:
If I appear to have suggested it should be linked to the window, then I
apologise - I mean nothing of the kind. Any reference I made was only to
indicate that there is code in existence that is able to get the data I'm
after, and an export to a code interface in addition to it being displayed
in that window is the aim of my PR.

Using an NCE connection seems fair enough, seeing as the analyzer is an NCE
product. A separate connection would also be fine, though - I have no idea
(being just a user) on the difficulties or niceties involved either way.

My aim is to use the data in code, which is why a scripting solution seems
less than ideal. Others, however, could easily access the same data stream
from a scripting environment should they want to do that, so that seems the
best of both worlds. That's assuming an export of the interface to the
Jython world.

I did not intend this to grow into a huge change to the existing window. If
there's a case to be made to modify the existing window to use the proposed
listener under the hood, possibly including the filtering options, then who
am I to complain? But it's not specifically part ofcwhat I'm looking for.

Wouter

vandoornw, 2025-05-25:
> Ah, my mistake. I didn't realize you meant Java code. My apologies for the
> confusion.
>
> Perhaps you can extend the existing class to do what you want:
>
> class ApiNcePacketMonitorPanel extends jmri.jmrix.ncemonitor.NcePacketMonitorPanel {
>
>     @Override
>     public void nextLine(String line) {
>         // Do new stuff here
>         // ....
>
>         // Then make sure the monitor processes it for e.g. display
>         super.nextLine(line);
>     }
>
>
>     /**
>      * Nested class to create one of these using old-style defaults
>      */
>     static public class Default extends jmri.jmrix.nce.swing.NceNamedPaneAction {
>
>         public Default() {
>             super("Open API NCE DCC Packet Analyzer",
>                     new jmri.util.swing.sdi.JmriJFrameInterface(),
>                     ApiNcePacketMonitorPanel.class.getName(),
>                     jmri.InstanceManager.getDefault(NceSystemConnectionMemo.class));
>         }
>     }
>
> }
>
> Then to invoke it:
>
>     var anpmp = new ApiNcePacketMonitorPanel.Default();
>     anpmp.actionPerformed(null);

Thanks, Bob, that looks like a very convenient way in. Unfortunately it'll
have to keep until the tourist season is over (November) before I have time
to put effort into it, but it certainly looks promising.

One more pointer would be good: will I need to have the Analyzer window
open for this to work?

Thanks again,
Wouter

vandoornw, 2025-05-28:
> My mistake. The method you're overriding has *two* String arguments:
>
>     public void nextLine(String line, String raw);
>
> The first is always displayed, and the second is shown if "show raw data"
> is checked. Not sure if that second is relevant to the NCE Packet Monitor
> or not.

Yay - that did it. Thanks!
Wouter

closed 2025-07-17 by bobjacobsen