JMRI/JMRI #14159: BUG? - recreating decoder index in 5.11.6

nigelcliffe, 2025-06-03:
I've just done an update from 5.9.3 to 5.11.6.  Machine is Windows10 (24Gb Ram), Java17.    There is different behaviour between 5.9.3 and 5.11.6 which may lead to confusion. 

Previous version 5.9.3 had just done an "update decoder index" and that worked correctly, introducing a new decoder in my user-space.   So, no issue with locally produced decoder file.  

Post update to 5.11.6: 
It happened that my installation would error on startup due to not having DCC system connected (Sprog on a Com port).  This is "normal for me", and I would just click "continue" to drop the system into a "not-connected, but I can edit stuff" state. 
The connections preference opened (normal) and the "recreate decoder index" also opened.  And that appeared to hang, doing nothing.  Progress bar didn't move, and after 8 minutes I cancelled the window.   No console errors reported (console below). 

I then closed the connection preference window, and manually ran "recreate decoder index".  This time, progress bar moved across, and completed in about 4 minutes.  But had failed to update, the new decoder (which appears in 5.9.3) did not appear, and a Java error is on the console (below). 

Re-ran "recreate decoder index" and no error, but decoder not introduced to "new loco" dialog. 

Re-started JMRI and decoder is now present.  

  

System Console: 

09:25:13,259 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
09:25:13,290 apps.util.Log4JUtil                   INFO  - This log is stored in file: C:\Users\Nigel\JMRI\log\session.log [main]
09:25:13,290 apps.util.Log4JUtil                   INFO  - This log is appended to file: C:\Users\Nigel\JMRI\log\messages.log [main]
09:25:13,321 apps.AppsBase                         INFO  - DecoderPro version 5.11.6+R1b48891fa4 starts under Java 17.0.11 on Windows 10 amd64 v10.0 at Tue Jun 03 09:25:13 BST 2025 [main]
09:25:16,098 apps.gui3.Apps3                       INFO  - Starting with profile Sprog_Programmer.3eb5e922 [main]
09:25:16,268 jmri.util.node.NodeIdentity           INFO  - Using b7281219-6883-4ec8-bba8-d1e5d193db27 as the JMRI storage identity for profile id 3eb5e922 [AWT-EventQueue-0]
09:25:16,432 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "SPROG Programmer" [main]
09:25:16,517 prog.serialdriver.SerialDriverAdapter ERROR - Serial port (none selected) not found: This port appears to have been shutdown or disconnected. [main]
09:25:16,519 prog.serialdriver.SerialDriverAdapter ERROR - failed to connect SPROG to (none selected) [main]
09:25:16,520 jmri.jmrix.ConnectionConfigManager    ERROR - Unable to create jmri.jmrix.sprog.sprog.configurexml.ConnectionConfigXml for [Element: <connection/>], load returned false [main]
09:25:16,522 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create several of your connections. [main]
09:25:16,955 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(C:\Users\Nigel\JMRI\Sprog_Programmer\roster.xml) [main]
09:25:17,017 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.util.startup.StartupActionsManager: jmri.util.prefs.InitializationException: Unable to run startup actions due to earlier failures. [main]
09:25:18,439 jmri.util.FileUtilSupport             INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
09:25:18,455 jmri.util.FileUtilSupport             INFO  - File path preference: is C:\Users\Nigel\JMRI\Sprog_Programmer\ [main]
09:25:18,455 jmri.util.FileUtilSupport             INFO  - File path profile: is C:\Users\Nigel\JMRI\Sprog_Programmer\ [main]
09:25:18,455 jmri.util.FileUtilSupport             INFO  - File path settings: is C:\Users\Nigel\JMRI\ [main]
09:25:18,455 jmri.util.FileUtilSupport             INFO  - File path home: is C:\Users\Nigel\ [main]
09:25:18,455 jmri.util.FileUtilSupport             INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
09:25:18,783 t.permission.DefaultPermissionManager INFO  - Permission file: C:\Users\Nigel\JMRI\.permissions.xml [main]
09:25:18,799 t.permission.DefaultPermissionManager INFO  - Permission system is enabled: no [main]
09:25:18,947 mri.jmrit.throttle.PowerManagerButton INFO  - No power manager instance found, panel not active [main]
09:25:22,331 ri.jmrit.decoderdefn.DecoderIndexFile INFO  - update decoder index [AWT-EventQueue-0]
09:33:48,917 ri.jmrit.decoderdefn.DecoderIndexFile INFO  - Decoder index recreation cancelled [decoderIndexer]
09:34:36,730 ri.jmrit.decoderdefn.DecoderIndexFile INFO  - update decoder index [AWT-EventQueue-0]
Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
09:40:20,128 ri.jmrit.decoderdefn.DecoderIndexFile INFO  - update decoder index [AWT-EventQueue-0]

danielb987, 2025-06-03:
@bobjacobsen 
Is this something for you to take a look at?

bobjacobsen, 2025-06-07:
When one installs a new JMRI release, there's a version number that's been incremented by the release process.  

When JMRI starts, it checks for a _local_ decoder index.  if it finds one, it checks the version number of that local index against the one in the release.  if the local one is smaller, it automatically remakes the decoder index, creating a new local one with the higher version number from the release.

This has the effect of only remaking the decoder index when it's needed to make sure any new decoder definitions in the release are included in the local decoder index.

Given the above, all that you saw was expected behavior _except_ that the remake decoder index failed when you proceeded with a partial, non-working system connection.  Everything else was as expected.  I've tried to reproduce that part with a failed SPROG connection without success.  

When the recreate-decoder-index process failed, what did the progress bar being "stuck" mean?  Was is stuck at 0%, or did it get to 100% and stall, or was it something else?

closed 2025-09-08 by github-actions[bot]