JMRI/JMRI #14167: Problem with EntryExit multipoint routing?

bobjacobsen, 2025-06-05:
TLDR:  I've found a case where entry-exit routing S->R and R->P (there is no Q) works, but S->P fails with an "Unable to create a multipoint route" Warning dialog box.  Not sure if it's a JMRI bug or a problem in the Layout Editor setup.  Help diagnosing this would be appreciated.  

I'm helping with a large club layout that uses JMRI's entry-exit capability to route across the layout.  In general, multipoint routing works fine:  When A->B and B->C are separately available, then you can directly request and get A -> C.

But in one specific case, it doesn't.  I haven't been able to figure out why.  And unfortunately I haven't been able to recreate this in a smaller version of the configuration.  

The .jmri profile is attached as a .zip file, as is the configuration file as .xml.txt. The behavior is seen on both the current HEAD of the master branch and JMRI 5.10 with Java 17. Signaling is done by script and SignalHeads, not SignalMasts.

The problem is this:  You can successfully request Sierra Main to Red Cliff main:

[image]

and you can successfully request Red Cliff main to Powderhorn Siding:

[image]

But if you try to go from Sierra Main to Powderhorn Siding, you get the Unable to create a multipoint route" Warning dialog:

[image]

All the other places I've tried, this kind of multiple-hop routing request works fine. In particular, Sierra Main to Powderhorn _Main_ works fine.  

I've gone over the Layout Editor connections and blocking carefully, and don't see anything wrong.  By process of elimination, it's beginning to look like a bug in the JMRI code.

When I repeat this with DEBUG logging enabled, I get:

```
     [java] 09:45:34,344 jmri.jmrit.entryexit.PointDetails     DEBUG - [nxButtonStateChange] Initial button or possible multiple point route [AWT-EventQueue-0]
     [java] 09:45:37,759 jmri.jmrit.entryexit.PointDetails     DEBUG - [nxButtonStateChange] Initial button or possible multiple point route [AWT-EventQueue-0]
     [java] 09:45:37,759 jmri.jmrit.entryexit.EntryExitPairs   DEBUG - [setMultiPointRoute] Start, from = NX-SierraM-EB, to = NX-PDS-EB [AWT-EventQueue-0]
     [java] 09:45:39,477 jmri.jmrit.entryexit.PointDetails     DEBUG - [nxButtonTimeOut] Flash timer cancelled [NX Button Timeout NX-SierraM-EB]
     [java] 09:45:39,477 jmri.jmrit.entryexit.PointDetails     DEBUG - [nxButtonTimeOut] Flash timer cancelled [NX Button Timeout NX-PDS-EB]
     [java] 09:45:39,478 jmri.jmrit.entryexit.EntryExitPairs   DEBUG - [setMultiPointRoute] Done, from = NX-SierraM-EB, to = NX-PDS-EB [AWT-EventQueue-0]
```

`NX-SierraM-EB` is the entry point sensor on Sierra Main; `NX-PDS-EB` is the desired exit point on Powderhorn siding.  So the last log line makes it sound like the routing worked. But I get that Warning dialing and the route isn't established.

Any suggestions on what to do next would be greatly appreciated.

[Dispatcher_Work_in_Progress.jmri.zip](https://github.com/user-attachments/files/20612384/Dispatcher_Work_in_Progress.jmri.zip)

[DispatcherDefault.xml.txt](https://github.com/user-attachments/files/20612394/DispatcherDefault.xml.txt)

bobjacobsen, 2025-06-05:
Thank you for looking into this!

When I replace the crossover 68 between main and siding with two turnouts, the problem goes away.  That's great and I can use that workaround.

Interestingly, if I just remove 68 entirely, closing up the main and siding tracks where it was removed, the "Autocreate Entry-Exit Pairs" tool never terminates. (Never say never, but I let it run for five minutes and it usually take a second or two) So there's something about this topology that's confusing the entry-exit code.

Looking at the block routing (below) not that there are two routes from Powerhorn Siding to Powderhorn Main:  One direct with a single count, through the crossover.  And one with _40_ hops and a score of 4250.  

Similarly from Powerderhorn Siding to Redcliff main: Two routes, one pretty direction and one that seems to loop around for a bunch of hops before getting there. This one is even more interesting, because it says that the route starting with Powderhorn Main (which I think means over the crossover) is the short one, and the one that starts by going directly to the P-R inter-siding track is the long one.

Finally, there aren't any pathological routes to Sierra, where all this started.

[image]

If I look at the routing from Sierra to Powderhorn or Red Cliff toward Powderhorn,  I don't see anything unusual.  The hop counts seem perfectly as expected/

Thanks again for your help with this, and for pointing me toward a workaround.

dsand47, 2025-06-05:
Using two turnouts instead of a crossover will change the logic paths.  

I did find the actual error.  The two crossovers in Yazoo each have a single block instead of an upper and lower block.  This caused a "reflection" in the route tables.

I added a single block that includes the crossovers and the connecting track segment.

[image]

Now the Sierra to Powderhorn siding works as expected.

bobjacobsen, 2025-06-07:
@dsand47 Thank you very much for your help with this!  

I was only looking between Sierra and Powderhorn as that's where I thought the problem was.  I didn't realize it could be further away. 

Was there a specific tool or technique you used to find the problem in Yazoo?

closed 2025-07-17 by bobjacobsen