JMRI/JMRI #14194: Differences in Store ALL from menu and when called on dirty Close

mstevetodd, 2025-06-16:
A user on Discord reporting an issue where new/changed Meter elements were not saved when answering Yes to "Do you want to Store ALL tables and panels?" Those same elements were saved properly if the user took the File, Store ALL table content and panels..." See screenshot.

[image]

Am I wrong to think these should call the same logic?

--SteveT

dsand47, 2025-06-16:
Based on a quick test, the meters are "disposed" early in the shutdown process before the store and load process runs.

dsand47, 2025-06-17:
> > Based on a quick test, the meters are "disposed" early in the shutdown process before the store and load process runs.
> 
> So, is there a suggested "fix" for this? Or are you saying this is expected?

The shutdown process has a number  general steps:

1. Run the "can shutdown" tasks.  Normally this is used for cleanup before shutdown but can also veto the shutdown.
2. Close windows.  This is where Meters is affected.
3. Run early shutdown tasks
4. Run the Store and Load checks
5. Run main shutdown tasks.

The issue with Meters is that the data disappears when the window goes away.  For other items, the data is separate from the window.

It might be possible to move the Store and Load checks before the window closing process.  That might be an issue if store and load is dependent on early shutdown tasks.

Another approach would be to separate the meter data from the meter presentation.

KenC57, 2025-06-18:
If the idea was that the compare test might abort the shutdown,
wouldn't that be a 'can shutdown' task? If it came up with a
difference, and the user said to abort the shutdown, like to review
the changes, it would say no to the 'can shutdown' query. At least
that's my limited understanding.

danielb987, 2025-06-18:
Yes. But @dsand47 wrote that the Store and load check isn't implemented as an ordinary ShutdownTask but is handled directly by the ShutdownManager.

By running the "can close" checks first and then run the Store and load test, you prevent showing the Store dialog when JMRI wouldn't quit anyway.