JMRI/JMRI #14200: DecoderPro Labels & Media: Settings for visibility of function buttons are lost when rebuilding the roster

andreasheckt, 2025-06-22:
Hi!

In 'Labels & Media', I can set the visibility of the function buttons:

[image]

After rebuilding the roster, the visibility settings (off-state) are lost:

[image]

In the corresponding roster's XML file, I was unable to find any information related to visibility. Do you have any ideas where this information might be stored?

I have installed JMRI 5.10+Rca461bd266 (Windows PC) and JMRI 5.12+Rc5bd588235 (Raspberry Pi), and the behaviour is the same on both installations.

Regards,
Andy

bobjacobsen, 2025-06-23:
> @bobjacobsen
> Do you have any idea of this issue?

I don't know much about these function labels at all.  I just tried setting some invisible in a sample roster.  I don't see anything to record them in either the `roster.xml` file or the specific roster entries XML file.  

And when I stop and restart the program, they all go back to having the visible box checked.  Does this feature really work in the expected way?

Lionel642, 2025-06-24:
just checked the code I contributed a while ago, I forgot to implement half of the stuff. Will push something.

andreasheckt, 2025-07-11:
Thanks for solving! 😀

commit c587b75f (2025-06-25): Fix #14200