JMRI/JMRI #14201: 5.12 Operations Pro problem;body=<!-- drag and drop the files in the open window onto here -->

Issue Type: __Bug__

I have just upgraded to 5.12 and now the program will not build trains. The status just says building but nothing is happening.

PanelPro 5

jskorch, 2025-06-22:
Will not build trains after upgrading to JMRI 5.12.

[session.log](https://github.com/user-attachments/files/20852362/session.log)

[systemInfo.txt](https://github.com/user-attachments/files/20852363/systemInfo.txt)

DanielBoudreau, 2025-06-23:
> @DanielBoudreau<https://github.com/DanielBoudreau>
> Is this something for you to take a look at?

Yes

DanielBoudreau, 2025-06-23:
jskorch,

Looks like you found a bug with the program.  Thanks for providing the session.log, it allowed me to quickly figure out what the problem is.  I believe you've manually assigned a car to a train that is sitting on a destination track.  The program incorrectly assumed that a car assigned to a train has a pickup time.  Could you please take a look at your Cars window and sort by train.  See if any of your cars have a train name and an asterisk.  An asterisk after a train name means the car has a manual train assignment.  If you only have a few, please remove the manual train selection and try building your trains again.  I'll have a fix in the next test release.

Dan

DanielBoudreau, 2025-06-23:
The other way to create this bug is to build a train using a prior version, and leave it built during the upgrade.  If this is your issue, reset all built trains and try building again.

DanielBoudreau, 2025-06-23:
> Dan,
> I reverted to the former version as I had an operating session today. I
> will not have another one until August because I'll be in Novi in July.
> This will give me plenty of time to reconfigure my trains and see if that
> solves the problem. Before I leave I will reset all the built trains and
> try again.
>
> Thanks, Joe

Joe,

What was your previous version?

Dan

jskorch, 2025-06-23:
I believe that it was 5.1*. I keep the most current version on my
operations computer.

Joe

jskorch, 2025-06-26:
Dan, I terminated the remaining trains then ran the program again and it
built the train just fine.
Thanks for your help on this.
Joe Skorch

DanielBoudreau, 2025-06-26:
Joe, thanks for reporting the problem.  Fixed in the next test release 5.13.1.

jskorch, 2025-06-29:
> Dan, I am not sure which files you want me to send to you.
>
> Joe

I don't know if this will offer any insight or not.

Joe

DanielBoudreau, 2025-06-29:
Joe,

Send the 6 operations files to my email address daboudreau@hotmail.com

here's a link to the help files for backups:

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#Backup

Dan

DanielBoudreau, 2025-06-29:
Joe, if you send me the session.log when the train build hangs, that might give me a clue to what is going on.

Dan

jskorch, 2025-06-29:
It would probably be a lot easier if you could call me at  (480) 558-6235.
There are so many options and I get confused on what to look for.
Joe

DanielBoudreau, 2025-06-29:
Same issue as previously reported. Closing issue.

closed 2025-06-29 by DanielBoudreau