JMRI/JMRI #14270: JMRI Permissions cause errors

dsand47, 2025-07-13:
After running JMRI 5.13.2, starting a previous version of JMRI causes errors:
```
15:52:21,654 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingTrack does not exists [main]
15:52:21,655 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingOnMain does not exists [main]
15:52:21,655 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingTrack does not exists [main]
15:52:21,656 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingOnMain does not exists [main]
15:52:21,656 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingTrack does not exists [main]
15:52:21,656 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingOnMain does not exists [main]
15:52:21,656 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingTrack does not exists [main]
15:52:21,656 t.permission.DefaultPermissionManager ERROR - Permission class jmri.PermissionsProgrammer$PermissionProgrammingOnMain does not exists [main]
```

danielb987, 2025-07-14:
Does this happen even if permissions are disabled?

I think it's safe to ignore this error. But it might be better to have it as a warning or info message.

dsand47, 2025-07-14:
> Does this happen even if permissions are disabled?
> 
> I think it's safe to ignore this error. But it might be better to have it as a warning or info message.

It is not possible to update 5.12 to change the errors.  

The easy fix is to delete the hidden file.  However, if Preferences are Saved regardless of the change, the hidden file comes back.

The final solution is to allow the file to be created using 5.12.  Then I made the file read only (`chmod 444` on macOS).  Subsequent preferences updates have an error message about the file but that is ok.

dsand47, 2025-07-14:
I recommend changing the messages to WARN in a future release.

This issue documents the potential issue for people who use multiple JMRI installs.  It also provides a technique to avoid the errors.

closed 2025-07-14 by dsand47

danielb987, 2025-07-14:
> I recommend changing the messages to WARN in a future release.

See PR #14275.