JMRI/JMRI #14288: Automatic save of changes to roster entry in DecoderPro

bobjacobsen, 2025-07-17:
(From the JMRI Community Meeting at the NMRA 2025 Convention)

Many modern applications save your work automatically as you make changes.  A user shut down his computer without storing in DecoderPro (so he didn't get the prompts to save).  It would be nice if DP could _optionally_ save to the roster entry after each change.

Smokifred, 2025-07-17:
May I suggest that this be optional. Many times I will try various values on the progrmming track (e.g. whistles) before wanting to save. My data is also stored on iCloud, so autosave may cause issues with updating the remote fiile.

Thanks for any consideration.

danielb987, 2025-09-01:
I think this issue should stay open.

closed 2025-11-24 by github-actions[bot]