JMRI/JMRI #14303: JMRI Web pages HW page updates

alain355, 2025-07-20:
Hi folks
On a french RR forum, a user asked if it was possible to update the names on the following JMRI web pages to include the YAMORC command station in the Dijikeijs (digirail) HW line. 
The Dijikeijs company is not existing anymore, but the designer of the initial DR5000 command station and accessories has created his own new company YAMORC (http://yamorc.de ) selling a new command station, accessories and also allowing owners of the initial DR5000 to update their DR5000 using a new chip with improved functions.

Communications between JMRI and either the  initial Dijikeisj DR5000, the updated YAMORC DR5000 or the new command station are the same than before.

Thus:
>> Replacement could be "Dijikeisj / YAMORC" on the main JMRI home page (https://www.jmri.org/)
>> And if possible update the products details on the product detail page (https://www.jmri.org/help/en/html/hardware/digirails/index.shtml) :
the various models are: DR5000 (old command station without update), YD9401 (update for DR000) and the new YD7010 (a complete new command station).

Thanks a lot.

Alain in Paris France

danielb987, 2025-07-27:
@alain355 @bobjacobsen 
I have tried to update the documentation by PR #14323. But documentation is not my strong side so I'm not sure if I have got it right.

danielb987, 2025-07-28:
> I've open the JMRI.org pages both from my computer (Win11 24R2) and my phone (Android) and both web pages (HW support and Dijikeisj details) are still the same, They don't look updated yet.

Yes, this is intended. When somebody wants to make a change to JMRI, he first creates a PR with the desired changes. Then somebody else reviews that and if everything is OK, the PR is merged. Changes to the web page happens after the PR is merged.

So before the PR gets merged, it needs to be reviewed. I wrote about my PR in this issue since I'm not entirely sure about what to change.

alain355, 2025-07-28:
Hi Daniel
So AFAIK, the way to update the pages isn't checked and validated. Thus they are not yet modified nor published?
***@***.***

bobjacobsen, 2025-07-28:
You can see the changes in the PR under the "Files Changed" link.  For the associated PR, this is at:

https://github.com/JMRI/JMRI/pull/14323/files

Ignore the formatting tags, just look for the text changes.

danielb987, 2025-07-29:
@alain355 
The PR is merged and the web pages are updated. I noticed one thing I have missed.

https://www.jmri.org/help/en/html/hardware/index.shtml
On this page, there is two places I haven't updated.

If you find anything more that needs to be updated, please tell me.

danielb987, 2025-07-29:
> I've published the info about the updated pages on the forum web page, and waiting if it pleases the requesters.

Thanks. I don't speek French so if they request changes, please write it here.

alain355, 2025-09-15:
OK to close, Daniel made the right change to the web pages.
Thanks

closed 2025-09-15 by alain355