JMRI/JMRI #14350: The effective LE level for track components changed by PR #14041

dsand47, 2025-08-09:
The track components are normally drawn at level 3.  The PR appears to have changed the effective layer to zero.

Reverting the change at line 485 fixes the problem.

Testing was done using Java 17.

See https://jmri-developers.groups.io/g/jmri/message/11398 for additional details.

devel-bobm, 2025-08-09:
Line 485 of which JMRI file?

dsand47, 2025-08-09:
> Line 485 of which JMRI file?

LayoutEditor.java.   See https://github.com/JMRI/JMRI/pull/14041/files

bobjacobsen, 2025-08-09:
This is an interesting bug.  At first, it looks like replacing an `Integer.valueOf(3)` with `3` would invoke auto-boxing if needed.

But in the `java.awt.Container` class, there are two methods:

```
Component  add​(Component comp, int index)
void add(Component comp, Object constraints)
```

The 2nd one is the desired one, where `constraints` sets the level of the display.  But changing to an explicit integer argument invokes the 1st one instead.  Since that exists, the compiler doesn't see the need to do auto-boxing to invoke the 2nd one.

danielb987, 2025-08-10:
@dsand47 
What happens if a user has an old panel file which he stores with 5.12 and then open and stores with my PR applied. Does the error then disappears or does the user needs to do something to fix it once the panel is stored with 5.12, even after my PR is applied?

Is this something that should be mentioned in the release notes of my PR?

dsand47, 2025-08-10:
> [@dsand47](https://github.com/dsand47) What happens if a user has an old panel file which he stores with 5.12 and then open and stores with my PR applied. Does the error then disappears or does the [...]

The track level is not stored in the xml file.  It is effectively a constant with a value of 3.

There are two cases where the error becomes visible.

The first is when a "background" image is added to a LE panel.  Background images are added at level 1 which hides the track which has defaulted to level 0 due to the error.  Most people set a background color which is not affected by the error.

The second case is when the level for an item is changed to less than 3 to have the track visible above the item.  This case is rare since everything else is normally at 4 or above.

The release note should be something about fixing a problem with LE track drawing and background images from 5.11.6 thru 5.13.3 including 5.12.  The 5.12 release note "New problems" section should also be updated.

devel-bobm, 2025-08-10:
So we do not need a script.

But should we plan for a 5.12.1, so that this is not an ongoing problem for those who _refuse_ to use a "Test Release"?

bobjacobsen, 2025-08-10:
I’ll update the release notes once a development build with the fix is available.

danielb987, 2025-08-10:
> The release note should be something about fixing a problem with LE track drawing and background images from 5.11.6 thru 5.13.3 including 5.12.

I have updated the release notes in PR #14352.

bobjacobsen, 2025-08-11:
See JMRI/website#570 for updates to release notes 5.11.6 through 5.13.3

closed 2025-08-11 by danielb987