JMRI/JMRI #14381: Engine Maintenance Scheduling

spark-fuel-support, 2025-08-24:
Can Engines have status, like load status for rolling stock?  I would like to send them for refueling, oil changes, repairs, etc.

DanielBoudreau, 2025-08-27:
Engines don't get the same respect as cars in OperationsPro!  To move locos, you use trains, which can be a local switcher. You would need to create a train for every move you wanted your locos to have.  For example, you could create a train to move a loco from the storage yard to your fuel or repair facility.  By requesting 0 car moves in the trains route, only the loco or locos would show movement on your Manifest or switch list.

Dan

spark-fuel-support, 2025-08-27:
Thanks.
   Could I add an engine refueling to the end of a route?  Typically your
train ends in Barstow, and drops all the cars into yards.  Would you
suggest making another location "Barstow Fuel", and making the engines wait
there for 30 minutes on the schedule, then returning to Barstow?  I was
thinking a route linked to the completion of the incoming route might work,
but I want to be sure it is the incoming engines that get serviced, and a
linked event might pick up other engines.

  To give an engine a bigger servicing, would you make a special route with
a very long delay at a location like "Barstow Engine Shop", and manually
assign the particular engine to it?

  Is there a way to list the engines currently at Barstow Fuel?

  Thanks for your help!

spark-fuel-support, 2025-08-27:
> You could add "Barstow Fuel" as a track at Barstow. Only allow a local
> train to pull from "Barstow Fuel" and return the engines to the storage
> track they normally reside in. You then build the local train to move the
> engines 30 minutes after the main train terminates. You could allow your
> main trains to pull directly from "Barstow Fuel" if that was okay, and they
> departed 30 minutes after the other trains were terminated.
>
> To force the engines to "Barstow Fuel", don't allow other trains to place
> engines at the engine storage track at Barstow, only the local.
>
> Dan

YES!!!  Your insight opens a whole world of ways to set things up.

So I can set up a local to take an available engine to the service shop on
Monday, and another one to return it on Friday.

Or, I could use an imaginary car type (called "Engine Service") which will
be delivered to the service shop, (a spur with a schedule for "Engine
Service", and limited routes to set out), and will stay, with the engine
(since that route cant leave engines anywhere else!) until the "Engine
Service" is complete.  A local route scheduled frequently for that spur
will notice that the "Engine Service" is available for pickup, and will use
that engine.  The "Engine Service" will be set out on a dedicated
fictitious spur (with a Random schedule) which will trigger another local
route (that can only use home power) to use an engine to bring it to the
service shop.

Powerful.

Can "Load" be used to filter routes, or trains, or engines?  I am thinking
that the "Engine Service" load might specify GP7, or SD40, or SW800, which
would then be able to trigger different times to complete the service.

Thanks!

*Dave Stewart*
Spark Contractors
1307 14th Ave
Dorothy, NJ 08317
(609) 476-2175
***@***.***
(609) 377-9711

closed 2025-09-25 by DanielBoudreau