JMRI/JMRI #14382: Random Results at Spurs.

spark-fuel-support, 2025-08-27:
I understand that the Random feature in the Spur Schedule can create more or less delivery demands, which is great.
  What I would like is, for each car which is spotted on the spur, a Random result.  For example, a custom load boxcar arrives.  Perhaps the dock hands don't clean it out well, or something spilled, or it is just Dirty.  So I want to be able to set up a 2% Random that the boxcar is shipped out as Dirty, and will route to a servicing track at the main yard.  And I can set a 1% Random for Damaged, which will send it to a repair facility.  then, maybe 30% or 50% it is refilled with a Custom Load, but it takes longer.  Perhaps some is given a defined Destination or pickup.  Whatever the rest of the 100% is will be handled in the last style, which would often be Empty.  Is there a way to do this?

danielb987, 2025-08-27:
@DanielBoudreau 
Is this something for you to take a look at?

spark-fuel-support, 2025-08-27:
> A spur with a schedule using the random feature should be able to do what
> you're requesting. Here's a schedule for car type "Boxcar" that pretty much
> does what you want. You would need to do this for every car type that you
> wanted to perform this particular operation. You could create a spur for
> each car type, apply a schedule to that spur, and pool all of the tracks
> together.
> image.png (view on web)
>
> Dan

Thanks for your response.
 So that will make a one car setout?   The online help instructions didn't
make that clear.  Maybe you could add your example to the help page.  It
seemed from the help that if there were 10 available cars somewhere on the
layout at 2% there would be a 20% chance of having one selected, and
delivered.  Perhaps it meant that if there was a 10 car setout there would
be a 20% chance that one delivered would be selected.

 So, for clarity, successive lines with Random and the following line with
no Random all refer to the same request?   How would I change this for 3
boxcars to be set out, with each one having its own Random result?
  What if there are two boxcars with different loads coming in, which need
a different Random table?

*Dave Stewart*
Spark Contractors
1307 14th Ave
Dorothy, NJ 08317
(609) 476-2175
***@***.***
(609) 377-9711

DanielBoudreau, 2025-08-28:
If during the train build process, you had available 100 boxcars that could go to the spur, then the odds are correct for that car type.  For example, 2% chance would mean that on average two boxcars would end up at that spur/schedule line if there were 100 cars available.  That's not 100 cars on the layout; all 100 cars need to be available to the build process.

If you had 10 cars available during the build process, the chance would only be 1/5 or close to none for one of the 10 cars to be selected by the schedule's line.

And if you had one car available, the odds are 1/50 that the 2% line would accept the boxcar.

Dan

DanielBoudreau, 2025-08-28:
Dave wrote:

"So, for clarity, successive lines with Random and the following line with
no Random all refer to the same request? How would I change this for 3
boxcars to be set out, with each one having its own Random result?
What if there are two boxcars with different loads coming in, which need
a different Random table?"

Not completely sure what you're asking, but here's an example for boxcars that the odds are tied to the boxcar's load:

[cid:2c398dbc-390b-4030-89e5-e8dc6a5dceaf]

So, if a boxcar with a load of "Nuts" hits this schedule, there's a 1% chance the car will go to the clean facility, a 5% chance to RIP, a 50% chance it will ship with Rocks, otherwise the car will get the E load.

Dan

spark-fuel-support, 2025-08-28:
Ok. Thanks for clarifying that.

  From our discussions I think you understand that I want the ability to
have one and only one car set out, but a Random final result.  I suppose
that by setting a spur length of 70 feet only one boxcar would fit.  Then I
could pool several spurs to represent the whole siding, each spur getting
one and only one, with a variety of outcomes.

Thanks!

*Dave Stewart*
Spark Contractors
1307 14th Ave
Dorothy, NJ 08317
(609) 476-2175
***@***.***
(609) 377-9711

closed 2025-09-25 by DanielBoudreau