JMRI/JMRI #14423: JMRI is not responding correctly to MQTT messages

jbasiago, 2025-09-17:
I have a device that sends messages to two channels; one for signals and one for text messages at roughly the same time.  Each device has a unique id but different topics for the two types of messages.  When a device sends a signal message, the corresponding signal is not changed.  I am receiving both signals in a script that registers to the two topics.  The odd thing is that the topic of the signal message is changed to be the same as the topic of the text message.  I believe this is why the signal is not changing state.  The signal messages are sent first, but there are a large number of these devices that reply from a single request, also provided by MQQT through JMRI.  This only seems to happen for this type of device and only for messages that are received.

danielb987, 2025-09-18:
@jerryg2003 
Do you have any idea on how to solve this issue?

jerryg2003, 2025-09-18:
I’d like to help but need more information.  What kind of “device” is sending the messages? [How many of these devices are there?] What are you using as your broker?  Is it on a Windows machine or a MAC?  Can you set up a message monitor (as Daniel suggests)?  One way of doing that on a Windows machine is to use mosquitto_sub.  We’ll really need to see a full list of all messages to figure out what is going on.  Thanks.

bobjacobsen, 2025-09-18:
What type of signals have you configured in JMRI?  Are you expecting them to automatically respond by themselves to receipt of an MQTT message, or do you have a script that's handling that?

closed 2025-09-18 by bobjacobsen