JMRI/JMRI #14427: PROGRAM NOT SAVING ROSTER ON SOME ENGINES

5284-dave, 2025-09-18:
Issue Type: __Bug__

I'm new to JMRI. I did save 3 engine entries but now the program is not saving 4 engine files to the JMRI file location. They will not go to the open roster location window

DecoderPro 5.12+Rc5bd588235
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.28+6-LTS mixed mode
Windows 10 10.0 amd64

bobjacobsen, 2025-09-18:
> I did save 3 engine entries but now the program is not saving 4 engine files to the JMRI file location. They will not go to the open roster location window

Could you give a little more detail on this please?  What does "they will not go to the open roster location window" mean? Do you get an error message?

5284-dave, 2025-09-18:
No error message. When I click save and move to another engine entry. The pictures disappear, all changes revert back to default in speed tables. Then I have to re-read the engine to get back the correct CV values. DO I have to also export the roster file?? I did that on the first few, but not the last four.